mne-tools/mne-python #6844: Improve mne anomymize

agramfort, 2019-09-30:
mne.anonymize should be doing similar things in cpp and python

cc @LorenzE

LorenzE, 2019-09-30:
@juangpc and @imsorryk are almost done working on a mne-cpp version of a fiff anonymization tool. PR can be found [here](https://github.com/mne-tools/mne-cpp/pull/393). I propose we try to finish the cpp version first and then port functionality to the existing mne-pyhton version. What do you guys think?

larsoner, 2019-09-30:
@bloyl expressed interest in working on this. I'm fine with waiting or if @bloyl wants to get started based on the concepts from the C code hopefully the final product wouldn't require too much adjustment

LorenzE, 2019-10-01:
@bloyl  and I will start to compare the current anonymized files from both mne-python and mne-cpp. I attached some slides ([mne_anonymize.pdf](https://github.com/mne-tools/mne-python/files/3676937/mne_anonymize.pdf)) by @juangpc and @imsorryk about the new mne_anonymize application. We are currently implementing solution 2. Tags which we are currently scanning for can be found in the switch case [here](https://github.com/mne-tools/mne-cpp/blob/dcc3875a15be93577515a96dc37befaf7133769b/applications/mne_anonymize/fiffanonymizer.cpp#L536).

agramfort, 2019-10-01:
@juangpc https://github.com/juangpc/fiff_anonymizer/blob/master/fiff_anonymizer.m should me moved to mne-matlab repo https://github.com/mne-tools/mne-matlab thanks for open a PR.

PR #6892 (2019-10-06): ENH: Basic anonymization
This modifies the way we were anonymizing meas_info objects to mainly permit time shifting dates.
It also adds an executable for anonymizing raw fif files from the command line.

The goal was to match the current plans in mne_cpp (https://github.com/mne-tools/mne-cpp/pull/393) which is still under development. hence the [WIP] tag.

closes  #6844 and #4982

@LorenzE
closed 2019-10-06 by larsoner