mne-tools/mne-python #6845: Remote visualization using dash and JS

SherazKhan, 2019-09-30:
# Problem
Remote visualization for brain surfaces and time series from fif files using data from S3.

# MVP:
A basic app that can load freesurfer segmented surface in the browser and overlay texture map.

GuillaumeFavelier, 2019-10-02:
I'm no expert in Dash but by adding a basic [dash backend](https://github.com/GuillaumeFavelier/mne-python/tree/dash_backend) I was able to obtain the following for the [`plot_source_alignment` example](https://mne.tools/stable/auto_tutorials/source-modeling/plot_source_alignment.html?highlight=plot_source_alignment):

[image]

[image]

[image]

**Note** The current PoC doesn't support color, transparency or glyphs.

agramfort, 2019-10-02:
that's nice to see you could pull this off so quickly. Do you think it's a
viable solution?

GuillaumeFavelier, 2019-10-02:
I would need to investigate further to check if Dash built-in features can follow the `_Renderer` contract to be honest.

agramfort, 2019-10-02:
ok. Let's see how far Sheraz can go with his approach

>

SherazKhan, 2019-10-04:
MVP:   https://github.com/mne-tools/mne-dash

closed 2019-10-04 by SherazKhan

drammock, 2019-10-08:
If we ever decide to build a more custom remote data browser, I just learned about https://en.wikipedia.org/wiki/VirtualGL which seems to solve one of the biggest challenges.