mne-tools/mne-python #6846: measurement Info shoudl have a _reprt_html method to be printed in a notebook

agramfort, 2019-09-30:
and the MNE report should be using the _repr_html method when available to produce HTML rendering of mne objects

larsoner, 2019-09-30:
I would wait to work on this until we can use https://github.com/sphinx-gallery/sphinx-gallery/issues/540 to see the outputs

agramfort, 2019-09-30:
this sphinx-gallery PR will help but it's quite orthogonal to me.

>

larsoner, 2020-08-31:
Anybody want a fun quick one for 0.21? @hoechenberger maybe?

hoechenberger, 2020-09-01:
> Anybody want a fun quick one for 0.21? @hoechenberger maybe?

Maybe I can squeeze it in somewhere, but no promises :)

vagechirkov, 2021-03-18:
```Python
import os
import mne

from IPython.display import display, HTML

sample_data_folder = mne.datasets.sample.data_path()
sample_data_raw_file = os.path.join(sample_data_folder, 'MEG', 'sample',
                                    'sample_audvis_filt-0-40_raw.fif')
raw = mne.io.read_raw_fif(sample_data_raw_file)

string = raw.info.__repr__()

template_red = '<span1 style="color: red;  margin-left: 150px;">{}</span1>'
template_blue = '<span1 style="color: blue;">{}</span1>'
info_html_string = '<div class="MathJax_Display" style="text-align: left; background-color:rgba(201, 76, 76, 0.3);">'
info_html_string += "<style> span1 {border-radius: 5px;  font-family:fantasy;} </style>"
info_html_string += '<span1 style="color: green; margin-left: 250px; font-size: large;">INFO</span1><br>'

for line in string.split('\n')[1:-1]:
    line_parts = line.split(':')
    for n, l in enumerate(line_parts):
        if n == 0:
            info_html_string += template_red.format(l + ':')
        else:
            info_html_string += template_blue.format(l)
    info_html_string += "<br>"

display(HTML(info_html_string))
```

@hoechenberger @agramfort is it something that you expect in the ```_repr_html``` method?

agramfort, 2021-03-18:
can you share a screen shot?

>

vagechirkov, 2021-03-18:
> can you share a screen shot?
> […](#)

The design is random :)

[image]

hoechenberger, 2021-03-18:
I'm wondering if we could "cast" the Info into a pandas DataFrame and simply use the DataFrame's HTML representation method?

agramfort, 2021-03-18:
pandas is not a hard dependency of mne

>

hoechenberger, 2021-03-18:
Ok. Then, how about repurposing what we already have in `Report`?
https://github.com/mne-tools/mne-python/blob/fc3f2b56bc172eb3b747751e636d55b0cbb56861/mne/report.py#L819-L874

agramfort, 2021-03-18:
+100 !

>

vagechirkov, 2021-03-18:
> Ok. Then, how about repurposing what we already have in `Report`?

Here is an example of ```raw_template``` output using ```IPython.display.HTML``` 

[image]

larsoner, 2021-03-18:
That's great! I think if the `html` is returned by `_repr_html_` it should work properly.

Incidentally you can show this off in our doc build by modifying some example that talks about meas info, and just having a line at the end of some code block like:
```
#############################################
# We can also get a nice HTML representation in IPython like:

info
```
Sphinx-gallery will recognize this statement-without-assignment and, if the object has a `_repr_html_`, add it to the output of the code block...

agramfort, 2021-03-18:
yes I would add _repr_html_ methods and then make sure that report.py just reuses these so we shoot 3 birds with one stone. It will work for the notebook, the doc and the Report

vagechirkov, 2021-03-19:
Here is am output of rendering `plot_report.py` tutorial:

[image]

closed 2022-06-15 by drammock