mne-tools/mne-python #6850: Adding Cortical Signal Supression (CSS) to MNE-python

johnsam7, 2019-09-30:
Add the cortical signal suppression (CSS) method as presented here: https://link.springer.com/article/10.1007/s10548-018-00694-5 

This is a method for studying subcortical structures, or parts of cortex in the medial surface of the cerebral hemispheres that are far away from the sensors. Based on the raw data, it removes the part of the signal subspace that comes from cortical sources while sparing subcortical data. Could be useful as a function on a raw object in the preprocessing stage.

larsoner, 2019-09-30:
Sounds good to me

jasmainak, 2019-10-01:
@johnsam7 don't hesitate to open a PR earlier than later so we can discuss on github how to proceed