mne-tools/mne-python #6854: restricting stc to label

jasmainak, 2019-10-01:
@agramfort this is the bug I was talking about today. Script to reproduce:

```py
import numpy as np

import mne
from mne.datasets import sample
from mne.minimum_norm import read_inverse_operator, apply_inverse

data_path = sample.data_path()
subjects_dir = data_path + '/subjects'
fname_inv = data_path + '/MEG/sample/sample_audvis-meg-oct-6-meg-inv.fif'
fname_raw = data_path + '/MEG/sample/sample_audvis_filt-0-40_raw.fif'
fname_event = data_path + '/MEG/sample/sample_audvis_filt-0-40_raw-eve.fif'
fname_label = data_path + '/MEG/sample/labels/Vis-lh.label'

event_id, tmin, tmax = 4, -0.2, 0.5
method = "dSPM"  # use dSPM method (could also be MNE or sLORETA)

# Load data
inverse_operator = read_inverse_operator(fname_inv)
raw = mne.io.read_raw_fif(fname_raw, preload=True)
events = mne.read_events(fname_event)

# pick MEG channels
picks = mne.pick_types(raw.info, meg=True, eeg=False, stim=False, eog=True,
                       exclude='bads')

# Read epochs
epochs = mne.Epochs(raw, events, event_id, tmin, tmax, picks=picks,
                    baseline=(None, 0), reject=dict(mag=4e-12, grad=4000e-13,
                                                    eog=150e-6))
evoked = epochs.average()

snr = 1.0  # use lower SNR for single epochs
lambda2 = 1.0 / snr ** 2

# Now, we generate seed time series by averaging the activity in the left
# visual corex
label = mne.read_label(fname_label)
src = inverse_operator['src']  # the source space used

stcs = apply_inverse(evoked, inverse_operator, lambda2, method,
                     pick_ori="normal", label=label) # label = None works
seed_ts = mne.extract_label_time_course(stcs, label, src, mode='mean_flip',
                                        verbose='error')
```

I get:

```py
Traceback (most recent call last):
  File "/Users/mainak/anaconda3/lib/python3.7/site-packages/IPython/core/interactiveshell.py", line 2719, in safe_execfile
    self.compile if shell_futures else None)
  File "/Users/mainak/anaconda3/lib/python3.7/site-packages/IPython/utils/py3compat.py", line 188, in execfile
    exec(compiler(f.read(), fname, 'exec'), glob, loc)
  File "/Users/mainak/Documents/github_repos/mne-python/label_bug.py", line 43, in <module>
    verbose='error')
  File "</Users/mainak/Documents/github_repos/mne-python/mne/externals/decorator.py:decorator-gen-217>", line 2, in extract_label_time_course
  File "/Users/mainak/Documents/github_repos/mne-python/mne/utils/_logging.py", line 89, in wrapper
    return function(*args, **kwargs)
  File "/Users/mainak/Documents/github_repos/mne-python/mne/source_estimate.py", line 2721, in extract_label_time_course
    label_tc = list(label_tc)
  File "/Users/mainak/Documents/github_repos/mne-python/mne/source_estimate.py", line 2609, in _gen_extract_label_time_course
    % (len(stc.vertices[i]), nvert[i]))
ValueError: stc not compatible with source space. stc has 13 time series but there are 3732 vertices in source space
```

larsoner, 2019-10-01:
This runs without error for me and should have been fixed by #6789, were you on latest `master` when you ran this code? Based on the traceback lines I don't think so

agramfort, 2019-10-01:
works for me on current master too

jasmainak, 2019-10-01:
cool, sorry about the noise :)

closed 2019-10-01 by jasmainak