mne-tools/mne-python #6862: BUG, VIZ: raw.plot_psd fmax param ignored when plotting

drammock, 2019-10-01:
the `fmax` parameter should be used to set the right-hand xlim of the plot, but it is not:

```
import os
import mne
sample_data_folder = mne.datasets.sample.data_path()
sample_data_raw_file = os.path.join(sample_data_folder, 'MEG', 'sample',
                                    'sample_audvis_raw.fif')
raw = mne.io.read_raw_fif(sample_data_raw_file)
raw.crop(0, 60).load_data() 

raw_downsampled = raw.copy().resample(sfreq=100)

for data in (raw, raw_downsampled):
    fig = data.plot_psd(fmax=350)
```
[image]

[image]

larsoner, 2019-10-01:
Somewhere there could be a `min(fmax, freqs[-1])` that makes it so that it uses the min of the actual freqs present and what you pass. But we can undo this / make it work.

agramfort, 2019-10-01:
I think it should rather crash if you ask for an fmax > sfreq / 2 as you go beyond nyquist cut-off frequency. There is nothing to see beyond.

PR #6879 (2019-10-03): FIX: Error plot psd fmax
Closes #6862
closed 2019-10-03 by larsoner