mne-tools/mne-python #6870: plot_psd with spatial_colors=True fails when some location values are nan

AKMeunier, 2019-10-01:
Adding a simple warning would help, it took me a while to realize that this was the reason my plot was all in black. Even better would be plotting only the nan electrodes in black and showind them beside the head in the little head color plot.

agramfort, 2019-10-01:
when trying I see in my terminal:

RuntimeWarning: Channel locations not available. Disabling spatial colors.

can you confirm that the warning is already there?

AKMeunier, 2019-10-03:
I only get these two:

.../site-packages/matplotlib/colors.py:259: RuntimeWarning: invalid value encountered in less if np.any((result < 0) | (result > 1)):

.../site-packages/matplotlib/colors.py:259: RuntimeWarning: invalid value encountered in greater np.any((result < 0) | (result > 1)):

agramfort, 2019-10-03:
please a file and a script so we can try to replicate

>

dtxe, 2022-04-22:
To replicate this issue:
1. Load any dataset with a DigMontage
2. Either use a dataset where one electrode has an undefined position OR manually set a sensor to nan coordinates:
```
from math import nan
montage = raw.get_montage()
montage.dig[1]['r'][0:2] = nan
raw.set_montage(montage)
```
3. Plot the evoked response with spatial_colors=True

PR #10559 (2022-09-16): ignore nan values when transforming 3d coords into colours
…for spatial_colors feature

#### Reference issue
Fixes #6870

#### What does this implement/fix?
Evoked butterfly plots with spatial_colors=True will plot all traces in black when any of the DigMontage coordinates are nan. The expected behaviour is that sensors with nan coordinates will be in black, while the remaining traces will be coloured.

This fix uses `np.nanmin` and `np.nanmax` instead of `np.amin` and `np.amax` when transforming x,y,z coordinates into colours, resulting in the expected behaviour.

#### Additional information
closed 2022-09-16 by larsoner