mne-tools/mne-python #6890: MNE Read_Montage

jgeller112, 2019-10-02:
I am trying to read in a .htps montage using the read_montage() function, but it looks to be broken. 

```
montage = mne.channels.read_montage(kind=montage1,unit='mm',transform=False) # read in average montage

Traceback (most recent call last):

montage = mne.channels.read_montage(kind=montage1(.htps monatge file)unit='mm',transform=False) # read in average montage

Attribute Error: module 'mne.channels' has no attribute 'read_montage'
```

With the newest version of MNE this should be deprecated not broken. And indeed it's missing from the 0.19 API:

https://mne.tools/stable/python_reference.html#module-mne.channels

agramfort, 2019-10-02:
use https://mne.tools/stable/generated/mne.channels.read_dig_hpts.html from
the new release.

if you use current master it's broken but if you use the 0.19 release it
should work with a deprecation warning.

>

jgeller112, 2019-10-02:
Using read_dig_hpts produces an error: 

n [294]: mne.channels.read_dig_hpts(montage1)
Traceback (most recent call last):

  File "<ipython-input-294-c5846a31f581>", line 1, in <module>
    mne.channels.read_dig_hpts(montage1)

  File "//anaconda3/lib/python3.7/site-packages/mne/channels/montage.py", line 493, in read_dig_hpts
    return make_dig_montage(ch_pos=ch_pos, **fid, hpi=hpi, hsp=hsp)

TypeError: make_dig_montage() got an unexpected keyword argument '2'

agramfort, 2019-10-02:
Can you share the file ?

Btw https://github.com/mne-tools/mne-python/blob/maint/0.19/mne/channels/__init__.py shows that read_montage is not removed from stable release

jgeller112, 2019-10-03:
I downgraded to .19 (was using .20) and things seem to work. Attached is the average montage I am using. 

[biosemi64_LocatorAdultsMedian.hpts.zip](https://github.com/mne-tools/mne-python/files/3683953/biosemi64_LocatorAdultsMedian.hpts.zip)

agramfort, 2019-10-03:
see https://github.com/mne-tools/mne-python/pull/6897

thanks for the bug report. We'll backport in 0.19 too.

>

PR #6897 (2019-10-03): fix reading of cardinals in .htps files
closes #6890
closed 2019-10-03 by larsoner