mne-tools/mne-python #6891: missing template for topoplot

cpelofi, 2019-10-02:
Hello, 
Trying to use evoked.plot_topomap on MEG data with .sqd format and receive this message error:

"The layout for the KIT system with ID 35 is missing. Please contact the developers about adding it."

Thanks for your help!
Claire

agramfort, 2019-10-02:
Can you share the file ? Where is the file coming from ?

cpelofi, 2019-10-02:
The file come from NYU MEG system.

File can be downloaded here:

https://we.tl/t-Ezmn5SI0lQ

Thanks,
Claire

teonbrooks, 2019-10-02:
it looks like christian updated the kit module to support that ID in `0.19` right before release. @cpelofi, which version of mne are you currently running? it might just need a quick update

cpelofi, 2019-10-03:
I just ran 

pip install mne --upgrade

Received that:

Found existing installation: mne 0.18.1
    Uninstalling mne-0.18.1:
      Successfully uninstalled mne-0.18.1
Successfully installed mne-0.19.0

Then ran again

evoked = epochs['tone'].average()
evoked.plot()
evoked.plot(spatial_colors = True, gfp = True)

times = np.arange(0., 0.4, 0.04)
evoked.plot_topomap(times=times)

And received the same message..

cpelofi, 2019-10-03:
I just upgraded to the 0.19 version but still get the same error message when I run the topomap.

agramfort, 2019-10-03:
@christianbrodbeck can you have a look?

christianbrodbeck, 2019-10-03:
Yeah I added sensor connectivity but not a layout. Is there a layout or a standard way for generating such a layout manually? Otherwise I would suggest to fall back on the automatic code in `find_layout` for missing KIT IDs. I can make PR if that seems like a good option.

agramfort, 2019-10-03:
whatever works. for MEG systems we tend to have predefined layout files.
Maybe fieldtrip has made one we can reuse?

>

christianbrodbeck, 2019-10-03:
I doubt it since this system has only just been updated (including a layout change)

closed 2019-10-04 by agramfort

teonbrooks, 2019-10-04:
@cpelofi, it looks like the fix was landed. it is on the development cycle so to use it, you will have to use this setup outlined [here](https://github.com/mne-tools/mne-python#get-the-latest-code)

cpelofi, 2019-10-04:
Working! Thanks

PR #6902 (2019-10-04): ENH _find_kit_layout():  add fallback on automatic layout
#### Reference issue
Example: Fixes #6891.

#### What does this implement/fix?
When no corresponding KIT layout is built in generate one automatically (previously raised an error).

#### Additional information
Leaves open the possibility to add manually built layout for any system later.