mne-tools/mne-python #6898: DOC: improvements to contributing guide

drammock, 2019-10-03:
This issue is to collect suggestions from Woods Hole code sprint participants about how to improve the contributing page. Add your comments below!

massich, 2019-10-03:
As a side note, I've been checking travis doc a lot these past 2 days and I found they have something really cool:
[image]

The button saying `Improve this page in github` which allows for easy PRs in the doc. Maybe we could add something similar.

AdoNunes, 2019-10-03:
Maybe add in the contribution page the requirement of adding the contributor's name and url in names.inc and explain how to write down the new contribution in latest.inc 

also, the link to what's new it's broken:
https://mne.tools/stable/install/whats_new.html

drammock, 2019-10-03:
@AdoNunes the link is https://mne.tools/stable/whats_new.html (stable) or https://mne.tools/dev/whats_new.html (dev)

massich, 2019-10-03:
adding that to the guide is good. Although IMHO a comment in a contributing guide is not enough to force people to add it. It should be a bot that forbids merging unless this info is there. But we should add it to the contributing guide none the less.

AdoNunes, 2019-10-03:
@drammock the broken link I referred to is at the bottom of the contributing webpage

drammock, 2019-10-03:
@AdoNunes Ahh, I see the problem now. Added a separate issue for that (#6901)

jasmainak, 2019-10-03:
Many folks seemed to have missed the part about forking. I think more pictures in the guide would really help. You can just reuse pictures from github (assuming that the URL stays stable) so that you don't add anything to the repo.

mohdsherif, 2019-10-03:
1- Make it more clear that a stable version under conda needs to have been installed (it took me a bit to figure I need this and that I cannot skip this step). 
2- For users on Linux who need to keep using their system python installation, note that they need to unset PYTHONPATH before installing.
3- Also, such users can switch back and forth when they need to use their system python by changing their PATH.

massich, 2019-10-08:
@jasmainak would it work to point to https://help.github.com/en/articles/fork-a-repo ?

drammock, 2019-10-08:
we already link to https://help.github.com/en/articles/fork-a-repo in the second paragraph of the contributing guide. So empirically, that was not sufficient.

jasmainak, 2019-10-08:
nobody clicks the links. The document is already long. What can we do to make someone up and running in less than an hour?

PR #7097 (2019-11-24): MRG, DOC: Update contrib guide
closes #6898 

This PR also updates the FAQ on filing new bugs, and the bug report issue template.

cc @mohdsherif @jasmainak @massich
closed 2019-11-24 by larsoner