mne-tools/mne-python #6905: plot_events throws ValueError if event_id contains additional entries

AKMeunier, 2019-10-03:
mne.viz.plot_events(events, event_id=event_id) produces 

ValueError: 17 from event_id is not present in events.

Is it possible to make this only a warning and let the code run through? In my experiments the stimuli are letters and not all letters are present in every session, and it is very tiresome creating a specific filtered event dict for every session.

agramfort, 2019-10-03:
ok with me

>

larsoner, 2019-10-03:
It would be good to replicate the `Epochs(..., on_missing=...)` API here

rderollepot, 2020-07-17:
Any news on this issue ?

If not, I can try to work on it as I am confronted with this problem too.

larsoner, 2020-07-17:
Nobody has worked on it as far as I know @rderollepot, feel free to give it a shot!

rderollepot, 2020-07-17:
If you don't mind I'd like to discuss the problem a little further, but I'd like to precise I'm very new to MNE and I'm using it to treat fNIRS data (working with @Swy7ch), so I might be missing a big part of the iceberg 😉 

First of all, in [this](https://github.com/mne-tools/mne-python/blob/aab9bae4cad04783d9457d2ae0edcbccf527e7b0/mne/viz/misc.py#L620) part of the code, _event_id_ is optional and its purpose is to add a legend to the plot if provided by the user (according to the function docstring).

The following piece of code (pointed out by the present issue) throw an error if you try to give a description in the legend to events that are not present in the dataset.
```
        for this_event in unique_events_id:
            if this_event not in unique_events:
                raise ValueError('%s from event_id is not present in events.'
                                 % this_event)
```
It seems to me that it should clearly not throw an error because here, _event_id_ just contains more information than required for this optional functionality. I wonder if this should even be a warning ? If we throw a warning, the purpose seems to be:
> Hey, are you sure your dataset is good, you don't seem to have all the events you want to describe ?

But is it the purpose of this optional functionality, in a plot function ? I'm not sure, and my preference goes to just skipping the unnecessary information. We would need however to filter _unique_events_id_ so it only contains event IDs present in the dataset to avoid having blank lines on the plot.

---

It seems to me that the piece of code coming just after is also problematic (should I open another issue ?):
```
        for this_event in unique_events:
            if this_event not in unique_events_id:
                warn('event %s missing from event_id will be ignored'
                     % this_event)
```
It is realized here that information is missing to completely update the legend and it is decided to ignore events that would have no entries in it. Thus, adding this optional parameters change what the plot contains, and it looks like an issue, isn't it ?

We could follow @larsoner suggestion and replicate the `Epochs(..., on_missing=...)` API to solve it, but I'm not sure what to do if the user select warning/ignore ? Partially update the legend, so we have legend entries following `event_label = '%s (%s)' % (event_id_rev[ev], count)` and others following `event_label = 'N=%d' % (count,)` ?

rob-luke, 2020-07-19:
I think it makes sense that if `on_missing='include'` (or something similar) then you would have a mix of text labels for the values defined in the dictionary, and then just trigger number then count as you describe for the others. And perhaps `on_missing=ignore` would just ignore the extra values?

I think this would be a handy feature to have and I would use it. Often when people send me data they collected it includes additional triggers to what they describe to me, so this would be a great time saver.

PR #8198 (2020-09-02): MRG, ENH: Add on_missing to plot_events
Closes #6905

@AKMeunier @rderollepot can you see if this works for you?

Regarding being able to ignore the existing warning, you can now do `verbose='error'` if you really want to also ignore the `event 2 missing from event_id will be ignored` warnings.
closed 2020-09-02 by larsoner