mne-tools/mne-python #6929: BUG: Misalignement plotting TFR with linear y-axis

stfnrpplngr, 2019-10-08:
### Describe the bug
Plotting TFRs with linear y-axis, bins for the particular frequencies are not properly aligned with the y-axis.

#### Steps and/or code to reproduce
[This gist](https://gist.github.com/stfnrpplngr/2378c70e6285322f38d54fa04d1fc764) is an example to produce the erroneous plot. The center of the notch filter appears not at 50 Hz regarding the plot.

#### Expected results
[image]
[image]

#### Actual results
[image]
[image]

#### Additional information
Platform:      Windows-10-10.0.16299-SP0
Python:        3.7.4 (default, Aug  9 2019, 18:34:13) [MSC v.1915 64 bit (AMD64)]
Executable:    C:\Continuum\Anaconda3\envs\general\python.exe
CPU:           Intel64 Family 6 Model 142 Stepping 10, GenuineIntel: 8 cores
Memory:        Unavailable (requires "psutil" package)
mne:           0.20.dev0
numpy:         1.16.5 {blas=mkl_rt, lapack=mkl_rt}
scipy:         1.3.1
matplotlib:    3.1.1 {backend=Qt5Agg}

sklearn:       0.21.3
numba:         Not found
nibabel:       Not found
cupy:          Not found
pandas:        0.25.1
dipy:          Not found
mayavi:        Not found
pyvista:       Not found
vtk:           Not found

stfnrpplngr, 2019-10-08:
The bin displaying the 50 Hz notch filter is not centered at 50 Hz.

larsoner, 2019-10-08:
@stfnrpplngr PR welcome!

agramfort, 2019-10-08:
@stfnrpplngr are you sure you did not paste twice the same image?

larsoner, 2019-10-08:
@agramfort look at the 90 tick they are in different places (both are wrong actually)

PR #6930 (2019-10-09): FIX: Aligning frequency bins to y-axis
#### Reference issue
Fixes #6929

#### What does this implement/fix?
Adjusting `extent` using `imshow` for linear y-scale. Now, frequency bins appear centered to the y-ticks.
closed 2019-10-09 by drammock