mne-tools/mne-python #6945: DOC: ambiguity in mne.pick_channels

c-torre, 2019-10-11:
Current documentation for mne.pick_channels states:

    Pick channels by names.

    Returns the indices of the good channels in ch_names.

Where good seems ambiguous and not defined.

https://mne.tools/stable/generated/mne.pick_channels.html?highlight=pick_channels

jasmainak, 2019-10-11:
A PR clarifying the docstring would be welcome. You would need to say that a good channel is something that is not in `info['bads']`

larsoner, 2019-10-11:
I don't think "good" should even be in the docstring here. It just picks indices of channels in `ch_names` that appear in `include` and not in `exclude`. If good and bad are even relevant to `include` or `exclude` -- or even reversed -- depends entirely on the user's use case.

jasmainak, 2019-10-12:
Agreed!

larsoner, 2020-03-04:
Fixed by #6957

closed 2020-03-04 by larsoner