mne-tools/mne-python #6947: improve plot_alignment to show 'dig' points from device coord frames

bloyl, 2019-10-11:
I'm routinely in the situation were I need to evaluate either the quality of the MEG HPI localizations and/or our external headshape measurements (Polhemus fastscan and fasttrack).

There currently isn't a way to visualize the MEG HPI locations since they are not in head coordinates or if you transform them to head coordinates, they typically end up labelled the same as the points coming from polhemus and you can't tell them apart.

My proposal would be to 
1) keep the points in the info['dig'] in the coord_frame they are measured in.
2) have plot_alignment move points from MEG coord frame to the display frame (HEAD or MRI) and plot them using a different glyphs (sphere:head_frame, ??:MEG FRAME)

I'm curious what others think of this or what problems it will cause :)

larsoner, 2019-10-11:
Some variant of this sounds workable to me.

Currently we allow showing fiducials from the head coord frame (`info['dig']`) and also in the MRI coordinate frame (from `subject/bem/subject-fiducials.fif`). The `info['dig']` ones use large translucent spheres and the MRI fids are smaller opaque ones.

It seems like we're in a similar situation here. Maybe the equivalent would be to use, unchanged, the opaqua small spheres for `info['dig']` HPI coils (since in this case this is our "ground truth") and use translucent larger spheres for the MEG coordinate frame HPI coils, and add a `meg_hpi_coils=False` default kwarg to `plot_alignment` (for your use case it would be `meg_hpi_coils=True`)?

neurogima, 2021-03-18:
I'll try with this one. I'm kind of familiar with the matter in matlab let's see how's in python

larsoner, 2021-03-18:
@neurogima a quick pointer is that currently you will probably find in the code that we cull the list of `dig` points to those in the head coordinate frame, and transforms those to the destination coordinate frame of the visualization (can be head, meg, or MRI). So I would think that the PR to add what @bloyl wants would loop over coord frames head/meg/MRI and make three separate lists of coordinates, and make sure each one (if not empty) gets transfromed and plotted in a different color.

larsoner, 2023-04-13:
You'd need to look at `plot_alignment` which IIRC under the hood uses some private function to plot dig points. Then understand a bit the coordinate transforms of MNE. It's not the easiest issue to tackle if you haven't thought about this stuff before. Maybe the EASY label is wrong here...