mne-tools/mne-python #6950: mne.channels.read_montage deprecated without deprecation cycle?

mmagnuski, 2019-10-11:
I might have missed something, but I have a feeling that `mne.channels.read_montage` did not raise any deprecation warnings before, and now it is unavailable. It took me a while to find out what I should use instead to create a standard montage. I could not find this information in whats new too (only the deprecation, but not what should be used instead). Was `read_montage` removed without a deprecation cycle?

larsoner, 2019-10-11:
It is in `0.19` but not `master`, so there is a deprecation cycle. It was deprecated right before release, though, so users of `master` barely saw any warning (just for a week or two).

You should use `make_standard_montage` or one of the dedicated `mne.channels.read_dig_*` readers

closed 2019-10-11 by larsoner

mmagnuski, 2019-10-11:
@larsoner Yes, I know what to use, I found it in one of the PRs. :) Good to know the deprecation is in 0.19, sorry for the noise.

larsoner, 2019-10-11:
Feel free to make a doc update for `maint/0.19`, though, the release notes and deprecation warning perhaps could have been clearer about what to use instead