mne-tools/mne-python #6953: Keep the sign values from TF-MxNE solver or irMxNE solver

dasdiptyajit, 2019-10-14:
Any possible way to keep the sign values (positive and negatives values) while performing source localization using TF-MxNE solver or irMxNE solver?

Note: Current solutions only provide absolute values in return as dipoles!

dasdiptyajit, 2019-10-14:
Follow up! @agramfort

agramfort, 2019-10-15:
do you think you could give it a try? it means adding a pick_ori parameter
like for apply_inverse and change the code accordingly.

>

dasdiptyajit, 2019-10-16:
yes, I have tired. I had to set a forward solution with fixed dipole orientations and had to modify the code and some parameters accordingly and it works nicely.

Note: One needs to flip the activated signals of a particular dipole after source localization, in order to match the signal strength across other dipoles due to the fixed orientation across hemispheres  i.e., left and right auditory cortex label.

dasdiptyajit, 2019-10-16:
Alright, I am closing the issue then!

closed 2019-10-16 by dasdiptyajit

agramfort, 2019-10-16:
ok great !

>