mne-tools/mne-python #6956: ModuleNotFoundError: No module named 'mne.viz.plot_sensors_connectivity'

MSA8D8, 2019-10-15:
I am faced with the problem below.

`import mne.viz.plot_sensors_connectivity`

> Traceback (most recent call last):
  File "D:\.....\~.py", line 24, in <module>
    import mne.viz.plot_sensors_connectivity as psc
ModuleNotFoundError: No module named 'mne.viz.plot_sensors_connectivity'
[Finished in 2.52s]

But, when I tried to `import mne.viz`, no error happened.
So, it seems to be something wrong with this plot_sensors_connectivity.

Thank you in advance.

cbrnr, 2019-10-15:
You need to do:

```
from mne.viz import plot_sensors_connectivity
```

closed 2019-10-15 by cbrnr

MSA8D8, 2019-10-15:
Thanks for reply, @cbrnr  

I tried your suggestion, but the situation is completely the same.

> Traceback (most recent call last):
  File "C:\....\~.py", line 25, in <module>
    from mne.viz import plot_sensors_connectivity
ImportError: cannot import name 'plot_sensors_connectivity' from 'mne.viz'

agramfort, 2019-10-15:
you must have an install problem or a too old mne version

>

MSA8D8, 2019-10-15:
@agramfort 

Thank you for comment.

I can solve the problem by update the version of mne.
Here is the `pip install -U mne` result.

> Installing collected packages: mne
  Found existing installation: mne 0.18.2
    Uninstalling mne-0.18.2:
      Successfully uninstalled mne-0.18.2
Successfully installed mne-0.19.0

Thank you both, @cbrnr and @agramfort

catalinasaini, 2024-07-04:
from mne_connectivity import plot_sensors_connectivity