mne-tools/mne-python #6965: Bug in envelopy_correlation()

ckiefer0, 2019-10-17:
Hi everyone,

I think there is a bug in the function`envelope_correlation()` in the case of `orthogonalize=False`. In the computation of the correlation, the mean is not being subtracted from `label_data_orth`.

By replacing line 110 in envelope.py with `label_data_orth = data_mag_nomean` this should be fixed.

agramfort, 2019-10-19:
thanks. Can you provide a small code snippet that demonstrates the problem?
how would you test that the result is correct after the fix? this helps us
writing tests to make sure the code does not get broken in the future.
thanks again

ckiefer0, 2019-10-21:
It seems like I was wrong. Both versions, the original code and my proposed "solution", give the same results. I calculated the correlation matrix manually and compared the results to the results of both versions of envelope_correlation(). You can find the code below.

I thought there might have been a bug because when one calculates the covariance in the numerator of the pearson correlation coefficient one usually subtracts the mean from both data series, X and Y. However, it is actually only necessary to subtract the mean from one of the time series.

`cov(X, Y) = <X - X_mean, Y - Y_mean> = <X, Y - Y_mean> - <X_mean, Y - Y_mean>`

Since '<X_mean, Y> = N * mean(X) * mean(Y) = <X_mean, Y_mean>' we have in the end:
`cov(X, Y) = <X, Y - Y_mean> = <X - X_mean, Y>`
with X, Y, X_mean, and Y_mean being vectors of length N, mean(X) and mean(Y) being scalars, and <,> being the dot product.

The code:
```
from mne.connectivity import envelope_correlation
from mne.filter import next_fast_len
import numpy as np
from scipy.stats import pearsonr
from scipy.signal import hilbert

n_epochs = 1
n_channels = 5
n_times = 100
data = np.random.rand(n_epochs, n_channels, n_times)
n_fft = next_fast_len(n_times)

data_hil = hilbert(data, N=n_fft, axis=-1)[..., :n_times]

corr = np.ones((n_epochs, n_channels, n_channels))

for epo_idx in range(data_hil.shape[0]):
    for idx1 in range(data_hil.shape[1]):

        data_mag = np.abs(data_hil[epo_idx])

        for idx2 in range(idx1 + 1, data_hil.shape[1]):

            corr[epo_idx][idx1][idx2] = pearsonr(data_mag[idx1], data_mag[idx2])[0]
            corr[epo_idx][idx2][idx1] = corr[epo_idx][idx1][idx2]

ec = envelope_correlation(data_hil, combine=None, orthogonalize=False)

np.allclose(corr, ec)
```