mne-tools/mne-python #6973: EEG locations not read in correctly

jasmainak, 2019-10-22:
It seems that for the Neuromag system we have here in Martinos, some folks have trouble reading in the `.fif` files. The locations of some EEG channels are not read in correctly.

See the code + data here to reproduce: https://www.dropbox.com/sh/l7erlqfy5dsezay/AABIMNHGD9KeUDPzdl2TiJuVa?dl=0

If you look carefully, the channel locations are in `raw.info['dig']` but not in `raw.info['chs']` for 4 channels. This issue has been [documented elsewhere](http://imaging.mrc-cbu.cam.ac.uk/meg/AnalyzingData/MNE_FixingFIFF).

The fix in MNE-C is to run the following:

```sh
mne_check_eeg_locations \
               —file <the  raw file you want to fix> \
               —fix
```

Seems to me that MNE-Python should handle this natively. Thoughts?

agramfort, 2019-10-23:
If I add this:

raw.set_channel_types({'EEG071': 'misc', 'EEG072': 'misc', 'EEG073':
'misc', 'EEG074': 'misc'})

it works. Why is there no location for these channels?

jasmainak, 2019-10-23:
that's not the right fix because those channels are not misc.

I updated the dropbox folder above with the file directly from the scanner (before any preprocessing). The issue is that in some Neuromag systems, the channels are ordered such that the first 60 channels are EEG, then you have 2 EOG and 2 ECG channels and finally you have the rest of the EEG channels. You can see that in `raw.info['dig']` all the positions are populated for 70 channels but this is not the case in `raw.info['chs'][idx]['locs']`

agramfort, 2019-10-25:
how about this?

```
import os
import mne

eeg_data_sj = './'
raw_data_file = 'awmrc_002_aw_1_raw.fif'

eeg_data_raw_file = os.path.join(eeg_data_sj, raw_data_file)

raw = mne.io.read_raw_fif(eeg_data_raw_file, verbose=False, preload=True)

picks = mne.pick_channels_regexp(raw.ch_names, regexp='EEG.')
ch_names = ['EEG000'] + [raw.ch_names[k] for k in picks]
montage = mne.channels.DigMontage(dig=raw.info['dig'], ch_names=ch_names)
raw.set_montage(montage)
raw.info['bads'] = ['EEG039']

eeg_data = raw.copy().pick_types(meg=False, eeg=True, exclude=[])
eeg_data_interp = eeg_data.copy().interpolate_bads(reset_bads=True)
```

jasmainak, 2019-10-25:
I think it fixes the problem -- I've asked for confirmation from those analyzing the data. Thanks! Anything to fix or document here?

agramfort, 2019-10-26:
if it's common need yes it should be documented somewhere or a dedicated
function should be added

>

jasmainak, 2019-10-28:
okay I see. Maybe @fmamashli could be motivated to make a PR? :)

agramfort, 2019-12-28:
anything to do here? close? update doc?

jasmainak, 2019-12-28:
let's close. I don't see a big push for this at the moment

closed 2019-12-28 by jasmainak