mne-tools/mne-python #6979: mne.time_frequency.tfr_array_morlet computes tfr for frequencies above Nyquist

gabri470, 2019-10-24:
#### Describe the bug
mne.time_frequency_tfr_array_morlets produces an output even when computing frequencies that violates Nyquist Theorem. 

#### Steps and/or code to reproduce
```Python
import mne
from matplotlib import pyplot

fname = mne.datasets.sample.data_path() + '/MEG/sample/sample_audvis_raw.fif'
raw = mne.io.read_raw_fif(fname)

tmp_data, time = raw[0:2,:] # data where sampled at 1000

d = mne.time_frequency.tfr_array_morlet(tmp_data[np.newaxis, ...],raw.info['sfreq'], [800], 5, verbose=False, n_jobs=16).squeeze()

pyplot.close()

f, (ax1,ax2) = pyplot.subplots(2,1,sharex=True)

ax1.plot(time, np.real(tmp_data[0,:].T))
ax1.plot(time, np.real(d[0,:].T))
ax1.set_xlim([20,22])

ax2.plot(time, np.real(tmp_data[1,:].T))
ax2.plot(time, np.real(d[1,:].T))

pyplot.show()

```

#### Expected results
It should rise an error for the violation of Nyquist theorem or at least a warning

#### Actual results
Instead it returns just the data unfiltered

larsoner, 2019-10-24:
Indeed it should check to see if any of the freqs are above Nyquist and raise an error. Are you up for making a PR to fix it @gabri470 ?

gabri470, 2019-10-24:
Dear Eric, 

I’d love to need to find time though. I’ll do my best,

gabri470, 2019-10-24:
Looking at the latest mne release I think this problem should be already been solved at lines 347 and above in `tfr.py/_compute_tfr` function

`    if (freqs > sfreq / 2.).any():
        raise ValueError('Cannot compute freq above Nyquist freq of the data '
                         '(%0.1f Hz), got %0.1f Hz'
                         % (sfreq / 2., freqs.max()))`

can't check for sure but at least it looks like it.