mne-tools/mne-python #6980: Coregistration rendering

stevenMgraham, 2019-10-24:
I am experiencing an issue with graphics/visualization when attempting coregistration. I am unable to visualize MRI subjects' heads. The GUI loads, allows for file import as well as displaying digitization points, but displays only a grey background. 

#### Various attempts at resolution
```Python
from mne.gui import coregistration
coregistration()
```
```Bash
$ python
$ mne coreg --simple-rendering
```
```Bash
$ export MESA_GL_VERSION_OVERRIDE=3.3
$ mne coreg
```
I have also tried the tips listed under [advanced installation of MNE-python](https://mne.tools/stable/install/advanced.html#troubleshooting-3d-plots-in-mne-python) to no avail.

#### Additional information (mne.sys_info(), $ glxinfo | grep version)
Platform:      Linux-3.10.0-957.1.3.el7.x86_64-x86_64-with-centos-7.6.1810-Core
Python:        3.7.4 (default, Aug 13 2019, 20:35:49)  [GCC 7.3.0]
Executable:    /local_mount/space/zazen/1/users/steven/anaconda3/bin/python
CPU:           x86_64: 12 cores
Memory:        31.2 GB

mne:           0.19.0
numpy:         1.17.2 {blas=mkl_rt, lapack=mkl_rt}
scipy:         1.3.1
matplotlib:    3.1.1 {backend=Qt5Agg}

sklearn:       0.21.3
numba:         0.45.1
nibabel:       2.5.1
cupy:          Not found
pandas:        0.25.2
dipy:          1.0.0
mayavi:        4.7.1 {qt_api=pyqt5, PyQt5=5.9.2}
pyvista:       0.22.4
vtk:           8.1.2

server glx version string: 1.4
client glx version string: 1.4
GLX version: 1.4
OpenGL core profile version string: 4.6.0 NVIDIA 410.79
OpenGL core profile shading language version string: 4.60 NVIDIA
OpenGL version string: 4.6.0 NVIDIA 410.79
OpenGL shading language version string: 4.60 NVIDIA
OpenGL ES profile version string: OpenGL ES 3.2 NVIDIA 410.79
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
    GL_EXT_shader_implicit_conversions, GL_EXT_shader_integer_mix,

larsoner, 2019-10-24:
Your graphics card should be fine so that shouldn't be the problem. Though if you could check that this code:
```
$ python -ic "from mayavi import mlab; mlab.test_surf()"
```
Produces:

[image]

It would help confirm that graphics are not the problem.

I suspect it might actually be a problem with your subject.

- Do you get any errors in the command line?
- Do you have this problem with the `sample` dataset? For example if you do `python -c "import mne; mne.datasets.sample.data_path(verbose=True)"` (warning: will download 1.5 GB to `~/mne_data/`) then `mne coreg -d ~/mne_data/MNE-sample-data/subjects -s sample` do you get this:

    [image]

stevenMgraham, 2019-10-24:
I am able to reproduce the figure from the Mayavi snippet. As for the sample data, I am able to reproduce that visualization in full as well.
I have not come across any command line errors when exploring the Coreg GUI except for when I view the Mayavi pipeline - I receive: 
```Bash
libpng warning: iCCP: known incorrect sRGB profile
```
though I do not believe that warning is so detrimental to be causing these issues.

larsoner, 2019-10-24:
Okay there is probably some issue with your head surface, then. It maybe was saved with the wrong units (?). Did you follow a standard FreeSurfer reconstruction then use `mne make_scalp_surfaces` or do something else?

stevenMgraham, 2019-10-24:
The reconstruction subjects directory I am trying to access was not processed by myself, and what is strange is that a different user on the very same machine I am now using is able to visualize the surfaces in the directory. 

I cross-checked a few libraries in their ```mne.sys_info()``` and found that our Python, vtk, mayavi, and mne versions were the same.

larsoner, 2019-10-24:
> The reconstruction subjects directory I am trying to access was not processed by myself

I wonder if it's actually a permissions issue. For example we might just check whether or not some files exist but never check if the user has permission to read them. Can you check to see if it is indeed some permissions problem? It would probably be `SUBJECT/bem/SUBJECT-head.fif`, `SUBJECT/bem/SUBJECT-head-dense.fif`, or `SUBJECT/surf/lh.seghead` that would be problematic

stevenMgraham, 2019-10-24:
I have checked my permissions belonging to the directory in question, and I do belong to the group that has the proper permissions. 

I have re-installed FreeSurfer to exhaust that possibility and in doing so noticed that I did not export a subjects directory in my ```~/.bashrc```.

I was shocked that _not_ explicitly exporting a subjects directory was the culprit! I had not done this previously and was simply loading the Coreg GUI, using the manila folder icon to direct me to the subjects directory for surface upload. 

```Bash
export SUBJECTS_DIR=/directory/of/interest # this was the missing link
source $FREESURFER_HOME/SetUpFreeSurfer.sh
```

Thank you very much for your help!

closed 2019-10-24 by stevenMgraham

jasmainak, 2019-10-24:
Probably related to: https://github.com/mne-tools/mne-python/issues/5204