mne-tools/mne-python #6988: subprocess env is not set properly

kingjr, 2019-10-28:
While running `mne.bem.mne_make_watershed_bem` I ran into a unspecific error:
`Error: io.UnsupportedOperation: fileno`

which is due to a subprocess error triggered earlier ('mri_watershed not found'). 

I tried changing FREESURFER_HOME via `mne.set_config` without luck.

I manually changed `run_subprocess`, as follow, which fixed the issue.

```python
def run_subprocess(command, return_code=False, verbose=None, *args, **kwargs):    
    import subprocess, os
    my_env = os.environ.copy()
    my_env["PATH"] = "/usr/sbin:/sbin:" + my_env["PATH"]

    cp = subprocess.run(command, 
                        env=my_env,
                        universal_newlines=True, 
                        stdout=subprocess.PIPE, 
                        stderr=subprocess.PIPE)
    return cp.stdout, cp.stderr
```

(And if someone opens a PR, you can pass the missing overwrite param here https://github.com/mne-tools/mne-python/blob/master/mne/bem.py#L1156)

agramfort, 2019-10-28:
you hacked the path?

this function looks very different from
https://github.com/mne-tools/mne-python/blob/master/mne/utils/misc.py#L85

are you using latest release/master?

kingjr, 2019-10-28:
> you hacked the path?

In the fix yes, else no.

> this function looks very different from

Yes, I went for the simplest possible fix ;)

This solved the issue for me, but I don't want to open a PR, because it's not a pure-python, platform-independent, easy-to-debug issue.

> are you using latest release/master?

yes

agramfort, 2019-10-28:
@larsoner any thought? It's hard to integrate a fix like this unless we can
replicate...

>

larsoner, 2019-10-28:
#6991 should fix it. Either use that branch, or `export FREESURFER_HOME=...` in your terminal *before* starting the Python interpreter (which is probably what most folks do, which is why nobody hit/reported this yet).

kingjr, 2019-10-28:
FWIW, I did export `FREESURFER_HOME` prior to python interpreter.

#6991 seems to works fine

closed 2019-10-28 by larsoner