mne-tools/mne-python #6994: add sprint2017 to mne-tools

teonbrooks, 2019-10-29:
back at the MNE sprint in Paris, I forked a copy of @kingjr's sprint2017 website with the intention of transferring ownership to mne-tools for posterity and historical artifacts. i don't have privileges to add repos to mne-tools, would someone with privileges like to make to fork the event so it can live here?

https://github.com/teonbrooks/sprint2017

larsoner, 2019-10-29:
> I forked a copy of @kingjr's sprint2017 website

You mean this?

https://github.com/kingjr/supervised_time_series

If so then the thing to do is probably to give @kingjr (temporary?) owner permissions to `mne-tools` so he can transfer the repository directly.

sappelhoff, 2020-01-22:
The original repo does not seem to exist anymore :-) so good that Teon forked it when it was still there.

teonbrooks, 2020-01-22:
@sappelhoff, just checked, the repo still exists but it's private on @kingjr's account. It would be a nice to have in the main repo for historical record but it currently seems stalled

larsoner, 2023-10-01:
Closing since I think we won't get around to this archival work :)

closed 2023-10-01 by larsoner

teonbrooks, 2023-10-02:
I kinda forgot about it. I transferred the fork of the repo to the mne-tools account since I never managed to get the original repo.