mne-tools/mne-python #6997: mne.events_from_annotations changes trigger codes

ibsneuro, 2019-10-30:
Hi all,
I have been reprocessing some EEG data acquired with a Neuroscan amplifier a few years ago and have noticed that the (now required) use of mne.events_from_annotations changes my trigger codes. Basic example:

`import mne

r = 'raw/26_EXP.cnt'

raw = mne.io.read_raw_cnt(r,
                          eog=['HEO','VEO'],
                          ecg=['EKG'],
                          emg=['EMG'],
                          preload=True,
                          verbose='DEBUG',
                          date_format='dd/mm/yy')

events, new_triggers = mne.events_from_annotations(raw)
`
My original trigger codes are still present in raw.annotations (67 and 173 are the two critical triggers in my case):

`list(raw.annotations)[0:5]`

> [OrderedDict([('onset', 36.774),
              ('duration', 0.0),
              ('description', '3'),
              ('orig_time', 1502689855.0)]),
 OrderedDict([('onset', 38.048),
              ('duration', 0.0),
              ('description', '2'),
              ('orig_time', 1502689855.0)]),
 OrderedDict([('onset', 39.589),
              ('duration', 0.0),
              ('description', '67'),
              ('orig_time', 1502689855.0)]),
 OrderedDict([('onset', 40.637),
              ('duration', 0.0),
              ('description', '173'),
              ('orig_time', 1502689855.0)]),
 OrderedDict([('onset', 64.553),
              ('duration', 0.0),
              ('description', '195'),
              ('orig_time', 1502689855.0)])]

But these are changed by the event extraction step (to 135 and 68, respectively):

`events[0:10]`

> array([[36774,     0,    94],
       [38048,     0,    75],
       [39589,     0,   135],
       [40637,     0,    68],
       [64553,     0,    71],
       [65980,     0,     1],
       [65995,     0,    72],
       [67559,     0,    53],
       [68577,     0,    80],
       [90069,     0,    71]])

I have seen that the correspondence between original and modified triggers is recorded in the dict produced by events_from_annotations (new_triggers in the above code) and that I will thus be able to reconstruct my trigger correspondences in this way. However, I'm wondering why this is the case and why it is necessary to have this extra complication. It is also not documented anywhere (as far as I could see) and took me some time to figure out; I thus expect that this is something that could lead to difficulties for a number of people in future.

Am I missing something here?

Many thanks in advance!
Ina

cbrnr, 2019-10-30:
In general, annotations can have arbitrary string descriptions so it is not possible to convert these to integer codes. Therefore, MNE by default assigns annotations to event codes starting from 1 (I believe) in steps of 1. In some special cases (notably BrainVision files), the function tries to extract numeric codes to create a more sensible mapping to integer event codes (see the docstring of [`mne.events_from_annotations`](https://mne.tools/stable/generated/mne.events_from_annotations.html#mne.events_from_annotations)).

In your case, you could pass a callable to the `event_id` parameter to specify a more suitable conversion, e.g. `event_id=lambda x: int(x)` or something along these lines. Note that this only works if all of your annotation descriptions can be converted to integers.

larsoner, 2019-10-30:
Agreed this seems like a case that should be taken care of by the `callable` version of `event_id`, so I'll close. But @ibsneuro let us know if it does not work for your use case and we can see if there is something to fix in MNE. Or if it was not clear from the docs feel free to make a PR to improve them or let us know what would have helped you find this option.

ibsneuro, 2019-10-31:
Thanks very much, @cbrnr and @larsoner! That indeed fixed my problem and saved me from constructing a more convoluted work-around. I think it would be useful to provide a bit more information in the docs -- I will have a think about this over the next couple of days and will either make a PR or send you my suggestions.