mne-tools/mne-python #7003: extract time series on volume source estimates

pshahbasak, 2019-10-30:
Hi-
I am using lcmv beamformer on resting-state MEG data using a 10mm grid. I will like to reduce the number of source time series using an anatomical atlas such as AAL. Is it possible to extract time series using MNE functions? I know this is possible for surface parcellations using read_labels_from_annot and extract_label_time_course. How do I go about doing this on volumetric source time series in MNE?

agramfort, 2019-10-31:
I would export to nifti and use
https://nilearn.github.io/modules/generated/nilearn.input_data.NiftiMasker.html

>

larsoner, 2019-10-31:
At some point we plan to tackle this, see discussion in #6155 -- closing this to keep discussion over there

closed 2019-10-31 by larsoner