mne-tools/mne-python #7006: Zoom to rectangle in raw.plot

cbrnr, 2019-10-31:
The `raw.plot` browser lacks a tool to zoom into a portion of the signal using the mouse. Would this be a useful feature? I'm willing to work on it (because I need it), but this will probably take some time to implement. Do you have suggestions on the UX? Maybe just trigger the action if the user drags the mouse (visual feedback in the form of a rectangle would be nice though)? Is there something similar already somewhere in MNE?

agramfort, 2019-10-31:
you mean change page duration? you can use this with keyboard shortcuts

>

cbrnr, 2019-10-31:
Yes, at least page duration. I know I can do it with keyboard shortcuts, but it would be faster if I could just drag the portion of the signal I'd like to zoom in with the mouse.

Zooming on the y-axis like that would also be nice, but I guess this won't be that easy/intuitive because of the way we currently display the individual channels.

agramfort, 2019-10-31:
> Yes, at least page duration. I know I can do it with keyboard shortcuts,
> but it would be faster if I could just drag the portion of the signal I'd
> like to zoom in with the mouse.
>
> Zooming on the [...]

brainstorm does this with rolling on the mouse with or without pressing
shift key AFAIK

UnResolvedExpression, 2023-01-09:
Hello @cbrnr.  Were you able to get this to work as you wished? I noticed on this page (https://mne.tools/stable/auto_tutorials/raw/40_visualize_raw.html) it is still suggesting to use hotkeys. I am interested in attempting to implement this myself and contributing to mne. Would you please advise on any potential challenges in this endeavor?

cbrnr, 2023-01-09:
This would be great @UnResolvedExpression! I think the main steps for implementing zoom to rectangle are:

1. Implement a new mode that can be toggled with a keyboard shortcut.
2. When that mode is active, the mouse cursor should change to indicate that users can now drag a rectangle.
3. After releasing the mouse button, get the left and right coordinates and set the visible signal to these limits.

One challenge for this feature is, I think, that is needs to work with both Matplotlib and Qt (PyQtGraph) backends. However, something similar is already implemented for marking annotations. You could take a look at how it's done and re-use parts of that implementation.

Finally, when we have zoom to rectangle, I think a "reset zoom" option would be very useful. This could e.g. always reset the visualization to show the first 10 seconds of data.