mne-tools/mne-python #7007: BUG: morph generated from src does not work

kingjr, 2019-10-31:
```python
# Prep example from scratch to make sure we have a unique pipeline

# Paths
data_path = sample.data_path()
subject = 'sample'
subjects_dir = os.path.join(data_path, 'subjects')
sample_dir = os.path.join(data_path, 'MEG', subject)

# Functional pipeline
raw_fname = sample_dir + '/sample_audvis_filt-0-40_raw.fif'
events_fname = sample_dir + '/sample_audvis_filt-0-40_raw-eve.fif'
raw = mne.io.read_raw_fif(raw_fname)
events = mne.read_events(events_fname)
epochs = mne.Epochs(raw, events, tmin=-.100, tmax=.200)
cov = mne.compute_covariance(epochs, tmax=0)
evo = epochs.average()

# Structural pipeline
trans_fname = sample_dir + '/sample_audvis_raw-trans.fif'
src = mne.setup_source_space(subject, spacing='oct6',
                             subjects_dir=subjects_dir, 
                             add_dist=False)
morph = mne.compute_source_morph(src,
                                 subject_to='fsaverage',
                                 subjects_dir=subjects_dir)
model = mne.make_bem_model(subject=subject, 
                           ico=4,
                           conductivity=(.3,),
                           subjects_dir=subjects_dir)
bem = mne.make_bem_solution(model)
fwd = mne.make_forward_solution(epochs.info, 
                                trans=trans_fname, 
                                src=src, bem=bem,
                                meg=True, eeg=False, mindist=5.0, n_jobs=-1)

# Inverse modeling
inv = make_inverse_operator(epochs.info, fwd, cov, loose=0., depth=0.8)
stc = apply_inverse(evo, inv)
# -----

```

critically
```python
# morphing from stc to stc works
morph = mne.compute_source_morph(stc,
                                 subject_from=subject,
                                 subject_to='fsaverage',
                                 subjects_dir=subjects_dir)

stc_success = morph.apply(stc)

# morphing from src to stc does not
morph = mne.compute_source_morph(src,
                                 subject_from=subject,
                                 subject_to='fsaverage',
                                 subjects_dir=subjects_dir)
stc_fails = morph.apply(stc)
```

returns 
```
~/opt/mne-python/mne/morph.py in _check_vertices_match(v1, v2, name)
   1160         raise ValueError('vertices do not match between morph (%s) '
   1161                          'and stc (%s) for the %s:\n%s\n%s'
-> 1162                          % (len(v1), len(v2), name, v1, v2))
   1163 
   1164 

ValueError: vertices do not match between morph (4098) and stc (3732) for the left hemisphere:
[    14     54     59 ... 155295 155323 155330]
[   841   1170   1329 ... 155249 155323 155330]
```

larsoner, 2019-10-31:
Does it work when you pass `fwd['src']` or `inv['src']`? It's possible/likely that the forward computation with `mindist` excluded some vertices

kingjr, 2019-10-31:
> Does it work when you pass fwd['src'] or inv['src']? It's possible/likely that the forward computation with mindist excluded some vertices

This works, thanks.

I wasn't aware the forward model could change the src. Shall I add a note in the tutorial?

larsoner, 2019-10-31:
That would be great, and the tutorial should use the one from fwd or inv, not the original (even if it's the same in that specific case, it helps show the principle)

agramfort, 2019-11-01:
>
> Shall I add a note in the tutorial?
>
+1

PR #7734 (2020-05-29): MRG, ENH: Add mixed source estimate support to compute_source_morph
Closes #7564.
Closes #7007.

@fleurgaudfernau can you see if it works for your use case? For example this code now runs fine:

<details>
<summary>Starter code</summary>

```Python
import mne
import os.path as op
import numpy as np

data_path = mne.datasets.testing.data_path()
subject = 'sample'
data_dir = op.join(data_path, 'MEG', subject)
subjects_dir = op.join(data_path, 'subjects')
bem_dir = op.join(subjects_dir, subject, 'bem')

fname_src = op.join(bem_dir, 'sample-oct-4-src.fif')
fname_aseg = op.join(subjects_dir, subject [... cut]
closed 2020-05-29 by agramfort