mne-tools/mne-python #7009: EEG-TMS Processing Example in MNE

padmasundaram, 2019-10-31:
Hi,

@Davi1990 and I would like to create an EEG-TMS example based on the EEG-TMS dataset provided in FieldTrip. @jasmainak suggested that we would need to get permission to fetch the data and use it in the example. There is interest in having a EEG-TMS example in MNE as several groups are trying to do similar preprocessing and analysis. So it would be good to have a example pipeline.

This is the FieldTrip example I am referring to:
http://www.fieldtriptoolbox.org/tutorial/tms-eeg/#dataset-information

agramfort, 2019-10-31:
@robertoostenveld is sitting next to me. He is fine with it but we need to make sure we give credit to fieldtrip and authors of the dataset (references etc). So +1

padmasundaram, 2019-10-31:
Thanks @robertoostenveld! Of course, we will credit field trip and the dataset authors/creators. 
I don't see a structural MRI for that dataset. Is it just an EEG data? I would like to have the full pipeline in the example with the inverse solution plotted on the cortex.

robertoostenveld, 2019-11-01:
The TMS-EEG tutorial does not include source reconstruction, it aims at cleaning and some simple channel level analyses. Afaik we also don't have the anatomical MRI of this subject. See http://www.fieldtriptoolbox.org/tutorial/tms-eeg/

padmasundaram, 2019-11-01:
We will have to use our own data for the source reconstruction example in that case.

agramfort, 2019-11-01:
would there be something specific about EEG+TMS for source imaging? Or once
I have cleaned up the sensor data I can just use any method I know? what
would we willing to demonstrate?

>

JonteP, 2023-05-26:
Did anything ever come of this? I would be very interested in such an example pipeline!

padmasundaram, 2023-05-26:
@JonteP : Is it enough to just remove artifacts from the EEG like in the FieldTrip example? Do you have a dataset in mind or do you want to just use FieldTrip's example?

JonteP, 2023-05-29:
Personally, I'm looking for an alternative to the TESA toolbox and pipeline
(https://nigelrogasch.gitbook.io/tesa-user-manual/example_pipelines) since
I'm migrating from Matlab to Python. But I am interested in anything to get
me started and I guess it needs to be general enough to serve as a general
TMS-EEG tutorial.

:)

jasmainak, 2023-05-29:
I agree that TESA is a high priority ... it's something many folks have asked for and certain parts are not too difficult to implement

padmasundaram, 2023-05-30:
I am open to working together on this. I would use the TMS-EEG application for my own project. Can we put this on a list for a code sprint?

Davi1990, 2023-05-30:
Hi @padmasundaram and @jasmainak! I am down to contribute. For preprocessing I use MATLAB and EEGLAB but I would be more happy to do all of the analyses in MNE tbh.

JonteP, 2023-05-31:
Glad to see there is interest in this! I cannot contribute much, except perhaps example data with our somewhat unusual equipment, if ever needed.