mne-tools/mne-python #7012: Read empty txt annotations

AdoNunes, 2019-11-02:
Hi

If the .txt annotation file is empty it gives a ValueError: not enough values to unpack (expected 3, got 0)

Sometimes, at least for me, it is useful to write empty files to keep track of steps already done. 
A simple option would be to add a try and except assigning empty values to the onset duration and desc.

What do you think?

This issue is similar to #6209

agramfort, 2019-11-03:
@AdoNunes please provide a basic code snippet we could use to replicate the error and make later a test. thanks

AdoNunes, 2019-11-03:
The snippet:

`from mne.annotations import Annotations,  _read_annotations_txt                                                                                                                                                            
Annot = Annotations([],[],[])                                                                                                                                                             
Annot.save('filename.txt')                                                                                                                                                                
_read_annotations_txt('filename.txt')    `
   
This seems to happen only for .txt

@agramfort in these cases, should I just do a PR directly or always first create an issue?

agramfort, 2019-11-03:
If you can do the PR it’s great 

Issue first is better

Thanks

PR #7014 (2019-11-04): FIX:no error if txt annotation empty 
  closes #7012

If saved annotation.txt is empty, when reading it it crashes, as it expects to get the onset, duration and description. 

Added a try and except ValueError. It does not crash, just gives a warning.
closed 2019-11-04 by larsoner