mne-tools/mne-python #7013: Env update test path conflicts

JavRodri, 2019-11-03:
HEllo, I'm new on this python/anaconda topics, please if this post is not in right place, let me know and I will correctit. 

I followed the instructions to install mne here: https://mne.tools/stable/install/mne_python.html
and then ran: conda env update --file environment.yml
 but i get next testpath conflicts and I honestly have no idea how to fix it, and I have been looking on internet for something that could help me out but so far nothing, could you please give me a hand?

UnsatisfiableError: The following specifications were found to be incompatible with each other:

Package testpath conflicts for:
anaconda==2019.10 -> nbconvert==5.6.0=py37_1 -> testpath
anaconda==2019.10 -> testpath==0.4.2[build='py37_0|py27_0|py36_0']
testpath[version='<0.4']
jupyter -> nbconvert -> testpath
spyder -> nbconvert -> testpath
_ipyw_jlab_nb_ext_conf -> jupyterlab -> notebook[version='>=4.3'] -> nbconvert -> testpath

larsoner, 2019-11-04:
What operating system are you on? It's weird that this happens because we test MacOS/Windows/Linux continuously with our instructions.

Did you very recently download Anaconda, or are you using an old version?

JavRodri, 2019-11-08:
Hello!

Im using win 10. and I downloaded the latest anaconda version and followed the procedure as shown in the link I provided in the first post. In the mean time I was looking other options and found a piece of advice telling me to install python 3 instead, the pip install mne tool and it seems to be working now. I do, however, would like to know what was wrong with anaconda procedure.

Thanks for following up.

drammock, 2019-11-08:
@larsoner do you know why we specify `testpath<0.4` in our env file?  Seems like that constraint may be the culprit; the second line looks like recent anaconda (`2019.10`) requires `testpath==0.4.2`

larsoner, 2019-11-09:
Probably some bug in testpath 4.0 and/or a library that uses it. If you look at blame:

https://github.com/mne-tools/mne-python/blame/master/environment.yml

It points to https://github.com/mne-tools/mne-python/pull/5585, a CI fix. I would remove the < 4 in a PR and see if everything works

PR #7044 (2019-11-10): MRG, MAINT: Remove testpath requirement
Let's see if everything works without this workaround. Probably things have been fixed in the last year to take care of whatever it was we were working around. See #7013 for related discussion.

Closes #7013.
closed 2019-11-10 by agramfort