mne-tools/mne-python #7015: mne.source_estimate.MixedSourceEstimate.plot_surface(...) crashes

christian-oreilly, 2019-11-04:
#### Steps and/or code to reproduce
Sorry for not providing a fully reproducible code, but I think the problem is quite clear here. I tried to plot sources with a MixedSourceEstimate using 
```
stc.plot_surface(views='lat', hemi='split', src=mne.SourceSpaces(src[:2]), subject='sample',
                             subjects_dir=atlas_dir, colorbar=False, clim=clim, colormap=colormap)
```
and I got:
```
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-72-14b84a8a0864> in <module>
---->   7                              subjects_dir=atlas_dir, #initial_time=max_t, colorbar=False, clim=clim, colormap=colormap)
     10 screenshot = brain.screenshot()
     11 brain.close()

~/mne-python/mne/source_estimate.py in plot_surface(self, src, subject, surface, hemi, colormap, time_label, smoothing_steps, transparent, alpha, time_viewer, config_opts, subjects_dir, figure, views, colorbar, clim)
   2240                                      config_opts=config_opts,
   2241                                      subjects_dir=subjects_dir, figure=figure,
-> 2242                                      views=views, colorbar=colorbar, clim=clim)
   2243 
   2244 

TypeError: plot_source_estimates() got an unexpected keyword argument 'config_opts'
```
And sure enough, on the current dev code at line 2257 of mne/source_estimate.py:
```
        return plot_source_estimates(stc, subject, surface=surface, hemi=hemi,
                                     colormap=colormap, time_label=time_label,
                                     smoothing_steps=smoothing_steps,
                                     transparent=transparent, alpha=alpha,
                                     time_viewer=time_viewer,
                                     config_opts=config_opts,
                                     subjects_dir=subjects_dir, figure=figure,
                                     views=views, colorbar=colorbar, clim=clim)
```
and at line 1540 of mne/viz/_3d.py
```
def plot_source_estimates(stc, subject=None, surface='inflated', hemi='lh',
                          colormap='auto', time_label='auto',
                          smoothing_steps=10, transparent=True, alpha=1.0,
                          time_viewer=False, subjects_dir=None, figure=None,
                          views='lat', colorbar=True, clim='auto',
                          cortex="classic", size=800, background="black",
                          foreground="white", initial_time=None,
                          time_unit='s', backend='auto', spacing='oct6',
                          title=None, verbose=None):
```
Unless I am missing something, this code is broken...

larsoner, 2019-11-04:
It probably has not been used for some time (and is obviously untested!). Are you up for making a PR to fix it?

christian-oreilly, 2019-11-04:
I can most probably have a look at it and if it goes well propose something. But I am on short deadlines for the next two weeks so it would go after that.

christian-oreilly, 2019-11-18:
Will be solved when PR #7079 passes. Closing the ticket.

PR #7079 (2019-12-14): Fix plot surface
#### Reference issue
Fixes #7015

#### What does this implement/fix?
Removed config_opts from the call to plot_source_estimates(). Some functionalities that were once served by config_opts may have been lost. I don't know what use case this parameter was previously fulfilling.
Fixed an issue with transparency. Now, if transparent has been set to None when calling plot_source_estimates(), it is set to True (the default for plot_source_estimates()) before calling brain.add_data(**kwargs).
Adding PySurfer in the dependencies.
closed 2019-11-18 by christian-oreilly