mne-tools/mne-python #7024: Error in raw.plot_psd() with bad segments

cbrnr, 2019-11-06:
I think #6748 broke PSD plotting when bad segments are present in annotations. The reason is that `psd_welch` now averages PSDs without taking care of NaNs, which it should because bad annotations are automatically transformed into `np.nan`.

```
import mne
from mne.io import read_raw_fif
from mne.datasets import sample
from mne.time_frequency import psd_welch

path = sample.data_path()
raw = read_raw_fif(path + "/MEG/sample/sample_audvis_filt-0-40_raw.fif")
picks = mne.pick_types(raw.info, meg=False, eeg=True)
raw.set_annotations(mne.Annotations(0, 10, "bad_segment"))
psds, freqs = psd_welch(raw, picks=picks)
```

In this example, all elements in `psds` are `np.nan`. Previously, when only mean was supported, we used `np.nanmean` to ignore NaNs. Not sure if there is an easy fix because `scipy.signal.welch` does not support NaNs AFAIK.

larsoner, 2019-11-06:
Sounds like a regression that needs a tiny test.

IIRC internally `welch` in SciPy just uses the STFT function. So we should just compute the STFT then do our own combining. We can compare the non-NaN version of our output with SciPy's. @cbrnr you up for this?

cbrnr, 2019-11-06:
> Sounds like a regression that needs a tiny test.

Definitely.

> IIRC internally welch in SciPy just uses the STFT function. So we should just compute the STFT then do our own combining. We can compare the non-NaN version of our output with SciPy's. @cbrnr you up for this?

Isn't that basically what we did before? We had only one option then (`np.nanmean`), but it should be pretty easy to implement the second option with `np.nanmedian`. So basically I'd revert most of the changes and just add the median option, right?

larsoner, 2019-11-06:
> Isn't that basically what we did before? 

I don't remember...

> I'd revert most of the changes and just add the median option, right?

I wouldn't automatically go with the revert route as it did add some functionality we want IIRC, such as new mode options and the ability to pass a callable, etc. (I think?). And there might be a cleaner way to do this than what we had before. But it's possible a revert would be cleanest, not sure.

Basically I would take the code as it is now work the way it should for data with NaN values as much as possible. Whichever way fixes NaN problems while keeping the newer flexibility the simplest way possible is fine by me.

We do have `mne.time_frequency.stft`, so I would personally try to use that and go from there.

cbrnr, 2019-11-06:
Previously, we used `scipy.signal.spectrogram` and then averaged the spectrograms with `np.nanmean`. Now we use `scipy.signal.welch` for both mean and median averaging (which doesn't support NaNs), but added an option to return the unaveraged spectrograms. IMO the simplest fix is the get rid of `scipy.signal.welch` again and just use `scipy.signal.spectrogram` (and implement or own averaging like before, but add `np.nanmedian`).

Not sure why we have `mne.time_frequency.stft` since this is also available in SciPy (we even have our own `fft` function in `fixes` so there might have been problems with the SciPy implementations at one point - maybe worth checking if this is still necessary).

larsoner, 2019-11-06:
STFT is from 0.19 and we only require 0.18

Agreed `scipy.signal.spectrogram` is a good option

cbrnr, 2019-11-06:
> STFT is from 0.19 and we only require 0.18

OK. I'll make a PR then using `scipy.signal.spectrogram`.

PR #7031 (2019-11-12): [MRG] Ignore NaNs when averaging spectrograms
Fixes #7024.

If this works, I'll add a test.
closed 2019-11-12 by larsoner