mne-tools/mne-python #7037: ENH: Add/modify mixin to allow setting info['meas_date']

larsoner, 2019-11-08:
On the mailing list people asked about modifying `info['meas_date']`. We should perhaps make a public method for this in cases where it's incorrect.

fabriceduprat, 2019-11-12:
When we create a raw object from a numpy array there is no absolute time recorded, it would be indeed  very interesting to be able to modify info['meas_date']. Thank you

agramfort, 2019-11-12:
if you have a datetime object `dt` in python you can do:

`raw.info['meas_date'] = (int(dt.timestamp() // 1), dt.microsecond)`

larsoner, 2019-11-12:
I was thinking of a nicer API like:
```
raw.set_meas_date(dt)
```
where `dt` must be a `datetime` object in UTC. It's a bit nicer than making people mess with `info['meas_date']` and also we can/should for example adjust the `raw.annotations` simultaneously when setting `info['meas_date']` to something new.

agramfort, 2019-11-12:
sure why not

>

fabriceduprat, 2019-11-13:
I tested Alexandre proposal, it works fine, a setter would effectively be nicer, but the same would be needed for other values like info['description'], perhaps propose a setter for all the values that we can modify ?
For info, why using the quotient (timestamp //1) ?

larsoner, 2019-11-13:
> I tested Alexandre proposal, it works fine, a setter would effectively be nicer, but the same would be needed for other values like info['description']

For `meas_date` we need the `setter` because there is something else that needs to be kept in sync with it, namely `annotations`.

For other things like `description` we don't need a setter as it's just a `str` entry.

We can make it clear in the `info` docs which things need to be set via setters and which do not. Most things shouldn't be modified by users so we can just make a short list of things that can be. For example `info['chs']` should not be modified, we have setters like `set_channel_names` and `set_channel_types` for that stuff.

agramfort, 2019-12-28:
we have now raw.set_meas_date and epochs.set_meas_date closing.

closed 2019-12-28 by agramfort