mne-tools/mne-python #7048: Implementation Robust Principal Component Analysis (RPCA) for EEG Noise Detection

rbechto2, 2019-11-11:
#### Describe the problem
Currently, MNE has no Robust Principal Component Analysis (RPCA) implementation and I think it would be a great addition to MNE's functionality. RPCA is a method to remove correlations from a data set, which works well with very corrupted data. RPCA is used to recover a low-rank matrix L and sparse matrix S from highly corrupted measurements, M = L + S. 

#### Describe your solution
To implement an RPCA function using the Alternating Direction Method of Multipliers (ADMM) optimization method in order to detect noise in a data set.
- Noise Detection 
- Noise Removal

#### Describe possible alternatives
There are different algorithms that can be used to implement RPCA optimization such as Augmented Lagrange Multiplier (ALM), Fast Alternating Minimization (FAM), or Iteratively Reweighted Least Squares (IRLS) method. 

@adam2392

agramfort, 2019-11-12:
is there published work that shows the benefit of this on EEG/MEG data?

>

rbechto2, 2019-11-18:
Hey @agramfort!

I found some great papers that demonstrate the benefits of RPCA on EEG data. The most common application is to use RPCA in order to improve tolerance to variability in EEG signals from trial to trial or day to day experiments and thus improving classification performance.

I also believe that this could be a useful function directly implemented to the mne package. I'd love to discuss this with you further. Thank you! 

Papers:
-"Improving Cross-Day EEG-Based Emotion Classification Using Robust Principal Component Analysis" (https://www.ncbi.nlm.nih.gov/pubmed/28769778)

-"Using Robust Principal Component Analysis to Reduce EEG Intra-Trial Variability" 
( https://www.ncbi.nlm.nih.gov/pubmed/30440781)

agramfort, 2019-11-18:
the way to proceed is to provide an implementation and an example using a
public dataset to show how useful this is.

>

Pranay144, 2019-11-27:
Does this issue still need work?
If yes, i can work on it.

adam2392, 2019-11-27:
@rbechto2 and our group at JHU is working on it within the context of pyautomagic and will link a demo when done. If the RPCA is general enough it would be awesome to just add into mne directly. 

Hope to have something by December preliminarily!

closed 2020-03-04 by larsoner