mne-tools/mne-python #7052: ENH: Store `info` dates as `datetime` objects

larsoner, 2019-11-12:
In principle there is no reason we can't store `info['meas_date']` and other date entries as `datetime` objects. @agramfort proposed this and it makes sense to me. On read we convert to `datetime` and on write we convert to what the FIF format wants to see.

I can work on it if people agree it's worthwhile.

cbrnr, 2019-11-12:
Good idea!

jasmainak, 2019-11-12:
what's the advantage of doing so?

larsoner, 2019-11-12:
`info['meas_date']` is easily interpretable by users, we hide the complications of the FIF format from them (they don't need to know the details)

jasmainak, 2019-11-12:
good idea!

should be done also for birthday?

larsoner, 2019-11-12:
If it falls under the "and other date entries" in the issue description I would say so. Basically anything we currently represent using tuple-of-int-as-date we should convert.

agramfort, 2019-12-28:
this is now done too.

closed 2019-12-28 by agramfort