mne-tools/mne-python #7053: Implementation of Robust Referencing Procedure as part of the PREP pipeline

Nick3151, 2019-11-12:
#### Describe the problem
Computing or selecting the reference signal is an important step when analyzing EEG recordings. It has been shown that both choosing one particular channel and using the overall signal average as the reference signal are problematic. The robust referencing procedure produces an estimation of the true average of the signal, i.e., if none of the channels are bad, and thus allows for subsequent processing steps of EEG signals such as detecting bad channels. To the best of my knowledge, there is currently no single function for computing robust reference in MNE.

#### Describe your solution
The robust_reference function iteratively detects and interpolates bad channels of EEG until it arrives at an average reference that is not affected by artifacts, as decribed in [PREP](https://www.frontiersin.org/articles/10.3389/fninf.2015.00016/full). The function will depend on find_noisy_channels, another function of PREP, which automatically detects bad channels in EEG.

#### Describe possible alternatives
In the future we'd like to implement the whole PREP pipeline for MNE.

@adam2392

agramfort, 2019-11-12:
maybe you can start from here https://github.com/sappelhoff/pyprep
by help stefan on this front?

>

larsoner, 2019-11-14:
Closing based on discussion in #7049

closed 2019-11-14 by larsoner