mne-tools/mne-python #7058: Explained variance of eLORETA

ezemikulan, 2019-11-14:
Hi, I've seen that the Explained variance reported in the logger when using eLORETA is very strange:

> Applying inverse operator to "stim"...
    Picked 62 channels from the data
    Computing inverse...
    Eigenleads need to be weighted ...
    Computing residual...
    __Explained -916356207643396166123520.0% variance__
    Combining the current components...

When using dSPM the value makes more sense (1.9%)

I've seen that here:

https://github.com/mne-tools/mne-python/blob/3fafe437413f29b91f36ca2726a4cceb87dfbab7/mne/minimum_norm/inverse.py#L906

>data_est_w.max()
Out[15]: 3070785165274.191
>data_w.max()
Out[16]: 21.2237224470818

Maybe it is related to https://github.com/mne-tools/mne-python/pull/6050?
I wish a could try a solution but it goes beyond my current knowledge about computing inverse solutions.

Cheers,

larsoner, 2019-11-14:
Yes it's currently wrong. It would help if we just printed "unknown" rather than something nonsensical. Would you be up for making a PR to do that when the method is eLORETA?

ezemikulan, 2019-11-14:
Sure, I'll do it. Thanks for the clarification.

PR #7061 (2019-11-15): add "Explained variance unknown" in logger for eLORETA
#### Reference issue
Fixes #7058

#### What does this implement/fix?
Logger displays "Explained variance unknown" when using eLORETA instead of an incorrect number.
closed 2019-11-14 by ezemikulan