mne-tools/mne-python #7067: meas_date is guessed during I/O when it is None

jasmainak, 2019-11-16:
It seems that the `meas_date` is not saved / read in correctly. There seems to be a regression of issue #4949 . Here is the code to reproduce:

```py
import os.path as op

import mne
from mne.datasets import testing

data_path = testing.data_path()
raw_fname = op.join(data_path, 'MEG', 'sample',
                    'sample_audvis_trunc_raw.fif')

raw = mne.io.read_raw_fif(raw_fname)
raw.info['meas_date'] = None
raw.save('temp.fif')

raw2 = mne.io.read_raw_fif('temp.fif')
assert raw2.info['meas_date'] is None
```

I discovered this while working on https://github.com/mne-tools/mne-bids/pull/290/

agramfort, 2019-11-16:
it was working with 0.19 ?

jasmainak, 2019-11-16:
Looks like @bloyl is the culprit :)

Git bisect tells me it stopped working after #5255:

```sh
1aea53a5e7b67ed82fa78c3fcf80d954bdf99a23 is the first bad commit
commit 1aea53a5e7b67ed82fa78c3fcf80d954bdf99a23
Author: Luke Bloy <luke.bloy@gmail.com>
Date:   Fri Jul 20 04:20:45 2018 -0400

    [MRG+1] CTF support for ds bad channels and bad segments (#5255)
    
    * CTF support for ds bad channels and bad segments
```

bloyl, 2019-11-17:
To be fair the guessing 
https://github.com/mne-tools/mne-python/blob/ef7d7c1899911b821312cc5be1558cc2d6ae4816/mne/io/meas_info.py#L1229-L1233
is from 2010. :)

All #5255 does is to ensure the `info['meas_id']` round trips through io.
https://github.com/mne-tools/mne-python/blob/ef7d7c1899911b821312cc5be1558cc2d6ae4816/mne/io/meas_info.py#L1300-L1302

So for a non anonymized info object (meas_date==None, but meas_id != DATE_NONE) the info doesn't round trip.

I think there are two reasonable fixes.
1) Always write a meas_date tag during IO. If `info['meas_date'] is None` then write `DATE_NONE`
2) During Reading don't guess`meas_date` from `meas_id`

jasmainak, 2019-11-17:
I agree with no guessing. Note that while it was broken before, didn't we fix it in #4990? Now looking at the diff I'm not so sure if we added a test to check that or prevent a future regression. Thanks for making the PR to fix it!

bloyl, 2019-11-17:
Potential Fixes

1. During Reading don't guess meas_date from meas_id. This is here.
https://github.com/mne-tools/mne-python/blob/ef7d7c1899911b821312cc5be1558cc2d6ae4816/mne/io/meas_info.py#L1229-L1230
I don't really know why this behavior was desirable but I'm sure it was/is for something.

2. Stop writing the `meas_id` directly in the `meas_info` tree of the fiff. This would revert to pre #5255 behavior. we'd also have to remove some of the testing that was added to #5255. This would work because somewhere in here:
https://github.com/mne-tools/mne-python/blob/ef7d7c1899911b821312cc5be1558cc2d6ae4816/mne/io/meas_info.py#L1211-L1224
the `meas_id` gets found and set to  to `DATE_NONE`

3. When writing the `meas_id` to the meas_info fiff tree, I could set the `meas_id = DATA_NONE` if `meas_date is None`. 

4. Do nothing. This issue indicates that `meas_date` and `meas_id` are out of sync (ie anonymized not through `anonymise_info`) and maybe its not surprising to get strange results?

I should note that I think that only case 1 would preserve round trip IO of both `meas_date` and `meas_id` in a situation where `meas_date = NONE` and `meas_id != DATE_NONE`. The other options will all guess one from the other in some way.

PR #7090 (2019-11-21): MAINT: Ensure that meas_date=None round-trips if `anonymize_info` is used
First step hopefully in our grand plan.

Closes #7071
Closes #7067
closed 2019-11-21 by agramfort