mne-tools/mne-python #7068: MNE Plot_Raw Showing Unit Label Conflict w/ YLabels

adam2392, 2019-11-16:
## Describe the problem
When I plot using raw.plot(), there is a line specifying units that conflict w/ the y labels.

See here.
[image]

#### Describe your solution
Is it possible to make this either on the other side of the plot, somewhere else altogether, or make it optional?

drammock, 2019-11-16:
You can turn that on and off. I forget off the top of my head what the key is, but if you push the "?" key while the plot is focused you'll get a list of all the keyboard shortcuts.

On November 16, 2019 9:59:32 AM PST, Adam Li <notifications@github.com> wrote:
>## Describe the problem
>When I plot using raw.plot(), there is a line specifying units that
>conflict w/ the y labels.
>
>See here.
>[image]
>
>#### Describe your solution
>Is it possible to make this either on the other side of the plot,
>somewhere else altogether, or make it optional?

adam2392, 2019-11-16:
I guess my issue is that I would like to see it and it is useful, but there is no way to put it somewhere else, or dynamically change it's size?

It's nice and out of the way when I have say only 5-10 channels showing, but when I have more, it gets in the way of the first channel text.

agramfort, 2019-11-17:
did you change manually the font sizes of matplotlib? another option is to
show less sensors it should expand on yaxis. Or we need to tweak to shift
if down a tiny bit. I don't see how to fix it with guarantees otherwise

>

cbrnr, 2019-11-18:
@adam2392 you can press "s" to toggle the scale bars. Currently, there is no argument to turn them off in the `raw.plot()` call though.

adam2392, 2019-11-19:
Okay no problem. Visualization is always tough to functionalize in my exp. 

I would say a programmatic option to turn off the unit scale would be helpful tho in general. If you think this is easy enough / okay to add in, I’d be willing to submit a PR.

If not, we can close this off.

drammock, 2019-11-19:
I would be OK with a `show_scale_bars` param (or maybe just `scale_bars`?) that defaults to `True` for backward compat.

cbrnr, 2019-11-19:
👍 for `scale_bars=True`, @adam2392 PR welcome!

PR #7091 (2019-11-21): FIX: Adding use_scalebars as a parameter.
Adds "use_scalebars" as a boolean parameter into plot() and plot_raw().

#### Reference issue
Example: Fixes #7068 

#### What does this implement/fix?
Added a boolean parameter to allow programmatic on/off of the unit scale bars.
closed 2019-11-21 by larsoner