mne-tools/mne-python #7073: [BUG] Dig point channel names do not match for ecog

alexrockhill, 2019-11-17:
```
import numpy as np
from scipy.io import loadmat

import mne
from mne.datasets import sample

data_path = sample.data_path()

mat = loadmat(mne.datasets.misc.data_path() + '/ecog/sample_ecog.mat')
ch_names = mat['ch_names'].tolist()
elec = mat['elec']  # electrode positions given in meters

montage = mne.channels.make_dig_montage(ch_pos=dict(zip(ch_names, elec)),
                                        coord_frame='head')

info = mne.create_info(ch_names, 1000., 'ecog', montage=montage)

raw = mne.io.RawArray(np.random.random((len(ch_names), 1000)), info)

print(raw.info['dig'])
print(raw.ch_names)
```

For some reason the dig channel names get set to EEG #X for dig points in ecog data rather than their given names in raw.ch_names.

jasmainak, 2019-11-18:
is this on `master` ?

alexrockhill, 2019-11-18:
Just checked on up-to-date master and can confirm it was off there.

agramfort, 2019-11-18:
dig points have no name.

In [3]: raw.info['dig'][0]
Out[3]: <DigPoint |     EEG #1 : (-13.2, -65.4, 70.2) mm   : head frame>

In [4]: raw.info['dig'][0].keys()
Out[4]: dict_keys(['r', 'ident', 'kind', 'coord_frame'])

the idea is that their index matches the order in raw.info['chs'] when it's
an EEG, iEEG or ECoG channel.

although their might be bugs

>

larsoner, 2020-03-04:
I think this is just a `__repr__` issue, the "EEG" just means that the kind of the dig point is `_KIND_EEG`. Maybe a doc update is needed?

larsoner, 2020-04-27:
@alexrockhill WDYT?

alexrockhill, 2020-04-27:
Sorry this dropped off the radar, I will look over this within the hour

alexrockhill, 2020-04-27:
> @alexrockhill WDYT?

Yes, looks like this is still an issue. I changed the example above slightly due to changes in montage. The output looks like this:

```
>>> print(raw.info['dig'])
[<DigPoint |     EEG #1 : (-13.2, -65.4, 70.2) mm   : head frame>, <DigPoint |     EEG #2 : (-14.5, -59.4, 78.1) mm   : head frame>, <DigPoint |     EEG #3 : (-15.6, -51.8, 83.8) mm   : head frame>, <DigPoint |     EEG #4 : (-15.4, -45.3, 89.5) mm   : head frame>, <DigPoint |     EEG #5 : (-14.9, -38.5, 94.0) mm   : head frame>, <DigPoint |     EEG #6 : (-13.1, -29.0, 98.5) mm   : head frame>, <DigPoint |     EEG #7 : (-13.6, -20.3, 100.9) mm  : head frame>, <DigPoint |     EEG #8 : (-10.9, -9.3, 103.5) mm   : head frame>, <DigPoint |     EEG #9 : (-21.5, -66.4, 69.1) mm   : head frame>, <DigPoint |    EEG #10 : (-23.3, -60.7, 76.9) mm   : head frame>, <DigPoint |    EEG #11 : (-24.7, -53.4, 82.2) mm   : head frame>, <DigPoint |    EEG #12 : (-26.3, -46.1, 87.1) mm   : head frame>, <DigPoint |    EEG #13 : (-26.1, -38.3, 91.8) mm   : head frame>, <DigPoint |    EEG #14 : (-25.3, -29.9, 94.9) mm   : head frame>, <DigPoint |    EEG #15 : (-23.3, -19.3, 98.9) mm   : head frame>, <DigPoint |    EEG #16 : (-21.1, -8.0, 102.6) mm   : head frame>, <DigPoint |    EEG #17 : (-30.4, -67.4, 67.8) mm   : head frame>, <DigPoint |    EEG #18 : (-32.8, -61.0, 73.8) mm   : head frame>, <DigPoint |    EEG #19 : (-33.7, -53.9, 79.3) mm   : head frame>, <DigPoint |    EEG #20 : (-34.4, -46.6, 84.2) mm   : head frame>, <DigPoint |    EEG #21 : (-36.3, -38.8, 87.8) mm   : head frame>, <DigPoint |    EEG #22 : (-34.5, -28.8, 92.6) mm   : head frame>, <DigPoint |    EEG #23 : (-33.5, -18.0, 95.5) mm   : head frame>, <DigPoint |    EEG #24 : (-29.9, -8.0, 97.7) mm    : head frame>, <DigPoint |    EEG #25 : (-39.5, -66.1, 63.2) mm   : head frame>, <DigPoint |    EEG #26 : (-41.7, -59.9, 69.0) mm   : head frame>, <DigPoint |    EEG #27 : (-42.6, -52.9, 74.4) mm   : head frame>, <DigPoint |    EEG #28 : (-43.8, -45.6, 79.5) mm   : head frame>, <DigPoint |    EEG #29 : (-45.7, -37.6, 83.0) mm   : head frame>, <DigPoint |    EEG #30 : (-44.6, -27.3, 88.1) mm   : head frame>, <DigPoint |    EEG #31 : (-43.0, -17.6, 90.1) mm   : head frame>, <DigPoint |    EEG #32 : (-39.6, -6.3, 88.4) mm    : head frame>, <DigPoint |    EEG #33 : (-44.6, -65.7, 58.1) mm   : head frame>, <DigPoint |    EEG #34 : (-48.2, -59.3, 63.2) mm   : head frame>, <DigPoint |    EEG #35 : (-50.5, -52.3, 69.4) mm   : head frame>, <DigPoint |    EEG #36 : (-53.6, -43.8, 73.1) mm   : head frame>, <DigPoint |    EEG #37 : (-54.8, -35.7, 76.9) mm   : head frame>, <DigPoint |    EEG #38 : (-53.3, -25.4, 80.8) mm   : head frame>, <DigPoint |    EEG #39 : (-51.2, -15.6, 82.3) mm   : head frame>, <DigPoint |    EEG #40 : (-47.3, -4.4, 80.8) mm    : head frame>, <DigPoint |    EEG #41 : (-51.8, -63.7, 51.3) mm   : head frame>, <DigPoint |    EEG #42 : (-55.0, -57.2, 57.4) mm   : head frame>, <DigPoint |    EEG #43 : (-57.5, -50.1, 62.5) mm   : head frame>, <DigPoint |    EEG #44 : (-60.6, -41.4, 66.6) mm   : head frame>, <DigPoint |    EEG #45 : (-61.4, -33.4, 70.9) mm   : head frame>, <DigPoint |    EEG #46 : (-61.2, -23.0, 73.3) mm   : head frame>, <DigPoint |    EEG #47 : (-58.9, -13.5, 74.3) mm   : head frame>, <DigPoint |    EEG #48 : (-55.3, -2.5, 73.9) mm    : head frame>, <DigPoint |    EEG #49 : (-57.5, -60.5, 45.5) mm   : head frame>, <DigPoint |    EEG #50 : (-61.5, -53.6, 50.0) mm   : head frame>, <DigPoint |    EEG #51 : (-63.5, -47.4, 55.4) mm   : head frame>, <DigPoint |    EEG #52 : (-66.0, -38.5, 58.4) mm   : head frame>, <DigPoint |    EEG #53 : (-67.1, -30.6, 63.0) mm   : head frame>, <DigPoint |    EEG #54 : (-67.2, -20.4, 64.4) mm   : head frame>, <DigPoint |    EEG #55 : (-64.3, -11.2, 63.8) mm   : head frame>, <DigPoint |    EEG #56 : (-60.2, -0.7, 64.0) mm    : head frame>, <DigPoint |    EEG #57 : (-61.4, -57.4, 37.2) mm   : head frame>, <DigPoint |    EEG #58 : (-65.9, -50.0, 41.2) mm   : head frame>, <DigPoint |    EEG #59 : (-68.1, -43.4, 46.3) mm   : head frame>, <DigPoint |    EEG #60 : (-69.4, -35.7, 50.3) mm   : head frame>, <DigPoint |    EEG #61 : (-70.4, -27.7, 53.0) mm   : head frame>, <DigPoint |    EEG #62 : (-70.5, -17.7, 54.1) mm   : head frame>, <DigPoint |    EEG #63 : (-68.0, -7.4, 53.0) mm    : head frame>, <DigPoint |    EEG #64 : (-64.2, 1.6, 53.9) mm     : head frame>]
>>> print(raw.ch_names)
['POL LPG1-Ref ', 'POL LPG2-Ref ', 'POL LPG3-Ref ', 'POL LPG4-Ref ', 'POL LPG5-Ref ', 'POL LPG6-Ref ', 'POL LPG7-Ref ', 'POL LPG8-Ref ', 'POL LPG9-Ref ', 'POL LPG10-Ref', 'POL LPG11-Ref', 'POL LPG12-Ref', 'POL LPG13-Ref', 'POL LPG14-Ref', 'POL LPG15-Ref', 'POL LPG16-Ref', 'POL LPG17-Ref', 'POL LPG18-Ref', 'POL LPG19-Ref', 'POL LPG20-Ref', 'POL LPG21-Ref', 'POL LPG22-Ref', 'POL LPG23-Ref', 'POL LPG24-Ref', 'POL LPG25-Ref', 'POL LPG26-Ref', 'POL LPG27-Ref', 'POL LPG28-Ref', 'POL LPG29-Ref', 'POL LPG30-Ref', 'POL LPG31-Ref', 'POL LPG32-Ref', 'POL LPG33-Ref', 'POL LPG34-Ref', 'POL LPG35-Ref', 'POL LPG36-Ref', 'POL LPG37-Ref', 'POL LPG38-Ref', 'POL LPG39-Ref', 'POL LPG40-Ref', 'POL LPG41-Ref', 'POL LPG42-Ref', 'POL LPG43-Ref', 'POL LPG44-Ref', 'POL LPG45-Ref', 'POL LPG46-Ref', 'POL LPG47-Ref', 'POL LPG48-Ref', 'POL LPG49-Ref', 'POL LPG50-Ref', 'POL LPG51-Ref', 'POL LPG52-Ref', 'POL LPG53-Ref', 'POL LPG54-Ref', 'POL LPG55-Ref', 'POL LPG56-Ref', 'POL LPG57-Ref', 'POL LPG58-Ref', 'POL LPG59-Ref', 'POL LPG60-Ref', 'POL LPG61-Ref', 'POL LPG62-Ref', 'POL LPG63-Ref', 'POL LPG64-Ref']
```

Where it seems not the best that they are labeled with EEG and a number. Any ideas about how to fix this?

larsoner, 2020-04-27:
Label them `EEG-like` or something else? digitization doesn't/can't know whether it's EEG, ECoG, sEEG, but most of the time it will be EEG.

alexrockhill, 2020-04-27:
What about just using the name of the channel?

larsoner, 2020-04-27:
`raw.info['dig']` does not know the channel names, it is just a list of dig points that can be cardinal, extra, or EEG type

alexrockhill, 2020-04-27:
Ok maybe nevermind then. How are the digitizations related to the voltage traces named by channel then out of curiosity?

larsoner, 2020-04-27:
The dig points can be used to assign `info['chs'][ii]['loc']` properly for EEG channels. But then you can pick/remove/add channels and `info['dig']` does not get updated, so there is no guarantee later that `info['dig']` has the same number or ordering of points as `info['chs']`.

alexrockhill, 2020-04-27:
Gotcha thanks for the explanation, seems like there must be a good reason for info['dig'] that is not captured by info['chs'][ii]['loc'] but I'm not sure it's clear to me what that is.

larsoner, 2020-04-27:
It can contain other points (headshape points, cardinal points) and EEG points no longer in the data. These are useful for coregistration even if you've `raw.pick_channels(raw.ch_names[:1])`'ed your data

alexrockhill, 2020-04-27:
Cool, okay well, I think that answers everything I thought might be wrong, thanks!

closed 2020-04-27 by alexrockhill

larsoner, 2020-04-27:
Where would we update documentation to make this clearer?