mne-tools/mne-python #7077: ENH: make mne python for CTF data preprocessing more friendly

AdoNunes, 2019-11-18:
@NataKozh and I have been doing some data preprocessing with CTF data and we found that mne-python needs a few PR upgrades to integrate common preprocessing steps done when maxfilter is not used. 

Namely:
- Muscle artifacts detection `detect_artifact_muscle`
   Detect data segments with muscle artifacts by bandpassing the data between 110-140 Hz, taking the envelope, z-scoreing and low pass filtering to smooth it. Then annotate the segments above a given threshold. This was written by @bloyl and is very similar to how fieldtrip detects muscle artifacts.

- Head movement detection and head position correction, `detect_artifact_movement` & `correct_head_pos`
  Detect segments where the head position is further away from a given threshold distance from the median head position during the recording. It iterates twice to find the median position during the segments with an acceptable head position. @bloyl was our source of inspiration and we are looking forward to his comments ;)
 
Another function that I would like to add:
- `Detect_bad_channels` 
It depends on the threshold used and it is useful to highlight the channels that might be bad, rather than an automatic step that does not require data visualization and inspection. It z-scores the data to find channel outliers based on power and/or does local correlation to find channels that do not capture similar magnetic fields as it's neighbour peers. 

I would like to add these functions and make an example tutorial. Let me know what do you think and if you have any suggestions. @jasmainak 

If you want to have a look at the functions we have them on: https://github.com/AdoNunes/ASD_MNE_pipe/blob/master/MNEprepro.py

agramfort, 2019-11-18:
I found some old code too
https://github.com/mne-tools/mne-python/blob/master/mne/preprocessing/bads.py#L8

it would be great if you break down this script in smaller functions.

if you find bad segment then return an Annotations with proper description
eg BAD_MUSCLE or BAD_MOVEMENT etc.

do we have some public data we could use to demo / validate this?

>

AdoNunes, 2019-11-18:
What about the brainstorm auditory dataset? There are bad channels and maybe the subject moved at least a bit...

I will call the `find_outliers` func from bads.

jasmainak, 2019-11-18:
what about making a separate repository for these issues as we did with `pyprep` in #7049?

agramfort, 2019-11-18:
At least start by an example so we see what function could be interesting for mne core library

AdoNunes, 2019-11-21:
I think that at least the muscle artifact and movement annotation would be good to have it in MNE. 

I have added a script example and the functions script on gist: https://gist.github.com/AdoNunes/b8edec72916543acdde716b6b23e926f

After computing the head position the head pos time series can be plotted (the horizontal line is the threshold, each line is the distance of one of the three cHPI to the median head position):
[image]

In this case, the participant moved slightly only during the brake

After computing muscle artifacts, the z-scores can be plotted:
[image]

and the annotations inspected:
[image]

agramfort, 2019-11-21:
could you open a PR with this code putting the function in an example for
now? So at least we can comment on it.

>

PR #7096 (2020-01-29): artifact detection
Closes #7077 

I have added (temporally in the wrong place) a function script to mark suspicious channels of being bad,  a muscle artifact detection (in practice almost the same as if Fieltrip) and movement annotation and head position trans correction.

The muscle and movement might be crucial in some datasets when Maxfilter is not applied and currently mne-python does not provide these functions.
closed 2020-01-29 by larsoner