mne-tools/mne-python #7081: [BUG] mne coreg GUI saves fiducials in meg space but labels as mri space

alexrockhill, 2019-11-19:
Try:

```
SUBJECT=sample
SUBJECTS_DIR=/path/to/mne_data/MNE-sample-data/subjects/
mne coreg
```

Save fiducials as test file

```
import mne
fids, coord_frame = mne.io.read_fiducials('/path/to/saved/mne_sample-fiducials.fif') 
print(fids)
```

The fiducials are in meg space but have kind 5 corresponding to head space.
```
[{'kind': 1, 'ident': 1, 'r': array([-0.07629625, -0.00062556, -0.00776012], dtype=float32), 'coord_frame': 5}, {'kind': 1, 'ident': 2, 'r': array([0.00267222, 0.09362256, 0.03224791], dtype=float32), 'coord_frame': 5}, {'kind': 1, 'ident': 3, 'r': array([ 0.07635873, -0.00258065, -0.01212903], dtype=float32), 'coord_frame': 5}]
```

Voxel space is something like:

```
array([[67.7471001 , 52.83904668, 48.1775196 ],
       [40.75665484, 24.14586344, 64.9023987 ],
       [16.50280451, 56.2472034 , 48.80732148]])
```

larsoner, 2019-11-19:
> Either the fids should be changed to voxel space or the coordinate system should be changed to head.

What we call `MRI` is the "Surface RAS" used by FreeSurfer documented here:

https://mne.tools/dev/overview/implementation.html?highlight=coordinate%20frames%20voxel#meg-eeg-and-mri-coordinate-systems

It's not the same as voxel coordinates. If you want to go from "MRI" to "MRI voxels", you'd want (the inverse of) the `img.header.get_vox2ras_tkr()` transform from `nibabel`, e.g.:

https://gist.github.com/larsoner/0ac6fad57e31cb2d9caa77350a9ff366#file-sample_mri_pos-py-L28

FYI that gist was used to get the electrodes from head coords onto the MRI for use in this example:

https://mne.tools/dev/auto_tutorials/source-modeling/plot_eeg_mri_coords.html

alexrockhill, 2019-11-19:
Very cool example, my mistake on the coordinate system. Looks good to me!

alexrockhill, 2019-11-19:
Ok sorry still a bit confused. 

I was using

```
mri_landmarks = apply_trans(trans, meg_landmarks, move=True) * 1e3
```
To get to what I thought was the mri space you are referencing.

Then
```
vox2ras_tkr = t1_mgh.header.get_vox2ras_tkr()
ras2vox_tkr = np.linalg.inv(vox2ras_tkr)
mri_landmarks = apply_trans(ras2vox_tkr, mri_landmarks)  # in vox
```
to get to voxels.

That worked for the landmarks saved out of the GUI, but skipping the first step did not.

i.e. voxel space
```
mri_voxel_landmarks = mne.channels.make_dig_montage(
    lpa=[66.08580, 51.33362, 46.52982],
    nasion=[41.87363, 32.24694, 74.55314],
    rpa=[17.23812, 53.08294, 47.01789],
    coord_frame='mri_voxel')
```

mri space

```
mri_landmarks = mne.channels.make_dig_montage(
        lpa=[-69.25741411, 10.58946009, -25.00085936],
        nasion=[3.37909386, 94.65942693, 32.25916435],
        rpa=[77.28562067, 12.05367187, -30.24882153],
        coord_frame='mri')
```

meg/head space
```
meg_landmarks = mne.channels.make_dig_montage(
    lpa=[-7.13766068e-02, 0.00000000e+00, 5.12227416e-09],
    nasion=[3.72529030e-09, 1.02605611e-01, 4.19095159e-09],
    rpa=[7.52676800e-02, 0.00000000e+00, 5.58793545e-09],
    coord_frame='head')
```

What I'm confused about is that the fiducials saved out of the mri gui are in meters (<<1) whereas the mri_space is related to voxels (>1). I'll keep working through it but if there's a simple explanation that would help a lot.

alexrockhill, 2019-11-19:
Ahhh I think I was just missing a conversion to mm

closed 2019-11-19 by alexrockhill

drammock, 2019-11-19:
@alexrockhill now that you understand it thoroughly, care to write a tutorial on coordinate transformations?

alexrockhill, 2019-11-19:
I could do that. Is there a tutorial for something else that it would be helpful if I modeled it off @drammock?

drammock, 2019-11-19:
Start here: https://mne.tools/dev/auto_tutorials/source-modeling/plot_source_alignment.html

See if it makes more sense to edit / improve that one, or if a separate stand-alone tutorial would be better.

alexrockhill, 2019-11-19:
Ok sorry super basic question: it looks like auto_tutorials are maybe in the .gitignore? How do I make a feature branch with this edited?

drammock, 2019-11-19:
the source of the tutorials are in `/path/to/mnepython/tutorials/`.  The rendered versions get put in `.../doc/auto_tutorials` by sphinx.