mne-tools/mne-python #7082: Wrong channel names in data set

jsosulski, 2019-11-19:
I have a dataset with 16 channels in which the channels `Fp1` and `Fp2` are unfortunately saved as `FP1` with a capital P. When I try to set_montage, mne reports that standard_1020 does not know the channels `FP1` and `FP2`.

My naive solution was to fix it by renaming the channels before setting the montage:

```python
for ch_i, ch in enumerate(raw_original.ch_names):
  if ch == 'FP1':
    raw_original.ch_names[ch_i] = 'Fp1'
  if ch == 'FP2':
    raw_original.ch_names[ch_i] = 'Fp2'
```

This leads to the error 

` info channel name inconsistency detected, please notify mne-python developers`

However, when I investigate the issue in the debugger and look at `raw_original.info['ch_names']` I can see that the channel names have also been modified by the above code.

Any solution to my issue? I do not own the data set, and as they are gdf files, I do not know how/if I can modify the binary blobs to fix the channel names.

cbrnr, 2019-11-19:
Can you try to use `raw.rename_channels({'FP1': 'Fp1', 'FP2': 'Fp2'})` before the `set_montage` call and see if this works?

jsosulski, 2019-11-19:
Yes this works! Thank you!

closed 2019-11-19 by jsosulski