mne-tools/mne-python #7092: Retain annotations from EDF files after concatenating

jeschj01, 2019-11-20:
If your issue is a usage question, please consider asking on our [Gitter channel](https://gitter.im/mne-tools/mne-python) or on our [mailing list](https://mail.nmr.mgh.harvard.edu/mailman/listinfo/mne_analysis) instead of opening an issue.

#### Describe the problem
Nicolet sometimes exports single .e files as multiple EDF files. When I go to load them into raw objects, annotations are often omitted because they are out of the data range. 

warning example:
<ipython-input-82-9a8378498a8d>:1: RuntimeWarning: Omitted 4 annotation(s) that were outside data range.

#### Describe your solution
I would like the option to retain annotations that are out of range, so I can then concatenate the multiple raw objects from multiple EDF files, and then have one raw object that has correct annotations.

#### Describe possible alternatives
Make it so omitted annotations are brought back after raw objects are concatenated making said annotations "in range" again.

#### Additional context
Add any other context or screenshots about the feature request here.

agramfort, 2019-11-20:
is an mne problem or an export problem?

we could also improv the nicolet reader
https://mne.tools/dev/generated/mne.io.read_raw_nicolet.html
to it supports annotations...

>

jeschj01, 2019-11-21:
An MNE fix would solve the problem, as the files still have the annotations, it's just that MNE is throwing them out when it sees that the annotations are out of range on the split up files. If i could just get the annotations before it decides to throw them out that would save me. Even if it gives indices for when the files are all put together, this is fine because I can just concatenate raw objects.

Brainstorm for example doesn't throw them out, but I'm not familiar enough to write code for that and want to avoid MATLAB if possible. This makes me think that there are indices from the original .e file header and that is what MNE is looking at, and then there are actual indices specific to the file that contains the annotations/events indices that Brainstorm is looking at. 

Unfortunately the Nicolet reader needs .data and .header files, and I'm starting with .e files.

Thank you

agramfort, 2019-11-21:
can you get them if you use
https://mne.tools/dev/generated/mne.read_annotations.html directly on the
edf files? maybe you can then use set_annotations to put them back on the
raw.

>

cbrnr, 2019-11-21:
EDF annotations can contain arbitrary values, no sanity checks are enforced by the file format. Therefore, MNE checks if annotations are within the time limits of the data.

If `mne.read_annotations` also throws invalid annotations away, we might think about adding an option to disable all checks and read all annotations.

jeschj01, 2019-11-21:
@agramfort I got so excited for this fix as it seemed to be working. Unfortunately  @cbrnr is correct and the actual "onsets" given for the annotation object are based on the time that the file started at, and in this case there are gaps in time (so if 4 hours of recording are omitted in the file and therefore relatively arbitrary values. 

Is there any function that will give indices for the annotation in regard to location in the file and not just based on the header/recording time info?  If not I will just have to go re-export these files and clip down to only the events that I want.

agramfort, 2019-11-21:
events that you can get with events_from_annotations function are given in
samples.
events are simple numpy arrays. You can also put back events in a fiel with
add_events method.

HTH

>

jeschj01, 2019-11-21:
@agramfort Unfortunately events_from_annotations requires a raw object, which when loaded omits my out of range annotations. If only there were a function that I can use with the edf file that gives the indices for annotations in samples.

cbrnr, 2019-11-22:
I don't quite understand your problem with onsets given relative to the recording start time.

In any case, you might want to try [EDFbrowser](https://www.teuniz.net/edfbrowser/) - I think this program displays all events (even if they are out of range), and you should be able to export these somehow.

jeschj01, 2019-11-22:
Sorry for the lack of clarity. 

I start out with 12 hour recordings. These are then clipped, so there are gaps in time. The start of of a file may be at noon, but then my event is at 4pm. It seems the header file gives annotation indices assuming I have a 12 hour file, as opposed to an 8 minute file. I think this is why my annotations are being omitted. Hence me wanting annotations that are specific to the number of samples between the beginning of the file and the annotation.

I'm going to try clipping my files so it's just around the event and hope this fixes it. Otherwise I may need to use another program like you say.

cbrnr, 2019-11-22:
Hm. Do you use `raw.crop` to clip your files? I thought this should correctly handle any annotations.

larsoner, 2020-03-04:
Closing due to non-response, happy to reopen if we get a response

closed 2020-03-04 by larsoner