mne-tools/mne-python #7093: creating custom montages without fiducials

nschawor, 2019-11-20:
Currently, it is possible to create custom montages with mne.channels.Montage. This now gives a
DeprecationWarning with the redirection to use DigMontage instead. 

As far as I understood it, this function requires reference points on the head, which I don't have. I only have standard coordinates for channels, for custom EEG caps. Below is the example use case which still works, but will not anymore in the next version. It would be great if this functionality would still be preserved. Or is it also currently possible with DigMontage somehow?

I also use the same type of workflow to create custom montages for working with ECoG grids, which also lack the reference points.

```python
import mne
import pandas as pd

df = pd.read_csv('channels.txt')
ch_names = df.name.to_list()
pos = df[['x','y','z']].values
montage = mne.channels.Montage(pos=pos, ch_names=ch_names, 
                               kind='motor-cap', selection=range(64))
montage.plot()
```
channel file: [channels.txt](https://github.com/mne-tools/mne-python/files/3871332/channels.txt)

Desired output:
[image]

agramfort, 2019-11-21:
did you have a look at new documentation?

https://mne.tools/stable/auto_tutorials/misc/plot_ecog.html?highlight=make_dig_montage
https://mne.tools/stable/auto_examples/visualization/plot_3d_to_2d.html?highlight=make_dig_montage
https://mne.tools/stable/auto_tutorials/source-modeling/plot_eeg_no_mri.html?highlight=make_standard_montage
https://mne.tools/stable/auto_examples/visualization/plot_montage.html?highlight=make_standard_montage

does it help?

>

nschawor, 2019-11-21:
Ah, I know understand that for ```mne.channels.make_dig_montage``` the parameter ```coord_frame``` has to be set to ```'head'```. Then it works. Thank you. 
If this is not the case, following ValueError is thrown, which made me assume that you need fiducials:
```python
    648     for pt in (nasion, lpa, rpa):
    649         if pt.ndim != 1 or len(pt) != 3:
--> 650             raise ValueError("Points have to be provided as one dimensional "
    651                              "arrays of length 3.")
ValueError: Points have to be provided as one dimensional arrays of length 3.
```

For this example:
```python
import mne
import pandas as pd

df = pd.read_csv('channels.txt')
ch_names = df.name.to_list()

pos = df[['x','y','z']].values
dig_ch_pos = dict(zip(ch_names,pos))
montage = mne.channels.make_dig_montage(ch_pos=dig_ch_pos)
montage.plot()
```

Maybe the documentation of the parameter for ```mne.channels.make_dig_montage``` could be extended to clarify this. :slightly_smiling_face:

agramfort, 2019-11-21:
@nschawor can you maybe give a try to a PR to help us with the documentation? thanks for your help and feedback

panixbbd, 2020-05-12:
[image]

codewithvicky, 2021-03-16:
Can i work on this issue please

codewithvicky, 2021-03-16:
sir can you help me in this because i am first time start contribution in open source code
please sir it is kind request

codewithvicky, 2021-03-16:
Or sir can you explain in more detail sir about this

drammock, 2021-03-16:
@codewithvicky fixing this issue will require some familiarity with neuroscience (electrode montages, etc).  I see from your profile that maybe you are just starting out learning python?  Perhaps our "EASY" label was misleading; the code changes are not too involved but the background knowledge required may be considerable.

codewithvicky, 2021-03-17:
ok sir

catalinamagalvan, 2021-03-18:
I'll work on this! :)

PR #9146 (2021-03-20): DOC: Add description about ´coord_frame´ parameter in mne.channels.make_dig_montage
#### Reference issue
Example: Fixes #7093.

#### What does this implement/fix?
Add notes about valid arguments for ´coord_frame´ parameter and its use for creating custom montages.
closed 2021-03-20 by drammock