mne-tools/mne-python #7094: Inconsistencies in plotting with layouts

jasmainak, 2019-11-21:
I'm working with CTF data where I don't have the channel locations and I would like to use the `layout` files as much as possible but I observe some inconsistencies in how it's implemented. Some of the topomap visualization functions include this as a parameter and others don't. For example if I do `evoked.plot_topomap` I can provide a `layout` but in the butterfly plot `evoked.plot` there is no option, so I can't see `spatial_colors` or get topographic plots of a selected region.

agramfort, 2019-11-21:
layout are supposed to be guess pretty easily for CTF data. DOes
find_layout work on your files?

>

larsoner, 2019-11-21:
>  Some of the topomap visualization functions include this as a parameter and others don't.

We are removing support for `layout` entirely in `topomap` examples. See long discussions in #7066, #5471, etc. Basically the idea is that Layouts are only used in topo-style plots (e.g., `plot_topo`) as a schematic, and never shown in relationship to a head outline. Any time you want sensors shown in relation to a head, a standardized routine flattens the sensor positions given in `info['chs'][ii]['loc'][:3]`.

So when you say:

> I'm working with CTF data where I don't have the channel locations

Do you mean it's not in `info['chs'][ii]['loc']`? If so then you should fix this, probably by using a montage if it's EEG data.

jasmainak, 2019-11-21:
Unfortunately it's not EEG data, so I can't use a montage. We don't have montages for CTF systems right? [Here](https://gist.github.com/jasmainak/8e9f3f7e229785789ddcbc073de5ed48) is a link to a (slightly longish) reproducible script that shows why I need `layout`.

larsoner, 2019-11-21:
I would manually set the `info['chs'][ii]['loc']` using some surrogate file. It's better and more future compatible than relying on layout.

agramfort, 2019-11-21:
@jasmainak you don't start from CTF data. Can't you read the fieldtrip with
our fieldtrip IO (mne.io.fieldtrip
https://mne.tools/stable/generated/mne.io.read_raw_fieldtrip.html)
if you don't have channel locations in your files it's hard to blame mne

>

jasmainak, 2019-11-21:
The FieldTrip IO is broken. I had to hack a solution ...

But I do need the layout to plot something. MNE doesn't have a gamma example and it's super important in neuro circles -- so I have to depend on FieldTrip unfortunately.

> I would manually set the info['chs'][ii]['loc'] using some surrogate file.

@larsoner I'm not sure I follow. The layout has 2D locations whereas the montage is 3D. How do you propose that I transform the 2D locations to 3D?

larsoner, 2019-11-21:
>  The layout has 2D locations whereas the montage is 3D. How do you propose that I transform the 2D locations to 3D?

By "surrogate file" I mean some CTF raw data file that actually has channel locations, not your layout file.

jasmainak, 2019-11-21:
okay looks like there is the channel information in the Fieldtrip data structure. I'll try to see if I can make it work.

But to check that the FieldTrip IO in MNE is broken, just do:

```py
read_epochs_fieldtrip('subjectK.mat', info=None, data_name='data_left')
```

where you can download this file from FieldTrip website: ftp://ftp.fieldtriptoolbox.org/pub/fieldtrip/tutorial/sensor_analysis/subjectK.mat

larsoner, 2020-03-04:
We killed a lot of the `layout` mechanics so this has hopefully been fixed, @jasmainak feel free to reopen if you still have problems with `master`

closed 2020-03-04 by larsoner

jasmainak, 2020-03-04:
sounds good @larsoner thanks

mmagnuski, 2020-03-09:
@larsoner @agramfort
Is `layout` considered for removal/deprecation alltogether? Working on #7404 (#7411) I found it a bit puzzling that `layout` does not work in `plot_topomap` (only in plot_topo functions). One nice thing about `layout` compared to `Info` is that is contains channel positions in a user-friendly way. It is also designed to hold 2d projection of channel positions - which makes sense for `plot_topomap`. Currently we have `DigMontage` and `Layout` and they seem similar enough that maybe they could be merged together? And if not maybe `layout` should work in `plot_topomap`?
I know this is only a minor nuisanse but it is a bit counter-intuitive that a class designed specifically for 2d channel projections (`Layout`) does not work in `plot_topomap` whose main purpose is to plot channel values projected to 2d. What do you think?

larsoner, 2020-03-09:
Layout is indeed no longer meant to be used in topomap, just topo. This was an intentional decision after some discussion.

mmagnuski, 2020-03-09:
Yes, I understand that (although I couldn't find the issues I mention here discussed there). So maybe a) layout should be removed and topo would use montage; b) or a cleaner approach : layout construction could accept sphere and layout would be used internally by plot_topomap (that is info -> layout if no layout is passed). Currently it wouldn't be very useful (so no need to do anything now) but would allow for easier customization of the topomap layout in some distant future (so maybe worth to keep in mind). :)

larsoner, 2020-03-09:
It's [here](https://github.com/mne-tools/mne-python/issues/3987#issuecomment-298920228) (and many linked issues, etc. from then if you want to follow the trail). The TL;DR is that if we show sensors in relation to a head circle we should do it in a single systematic way, i.e., via our topomap_to_sphere functions. Layouts are only to be used in cases where a head sphere is not drawn (e.g., selecting sensors in `raw.plot`, in `plot_topo`, etc.).

> but would allow for easier customization of the topomap layout in some distant future (so maybe worth to keep in mind).

This is actually what we're trying to avoid. If you want to mess with your sensor positions during plotting you should not be using `plot_topomap`, which (in principle) is meant to make use of a systematic relationship/transformation between actual sensor positions and the subject's head.

mmagnuski, 2020-03-09:
Thanks for the link, I even upvoted that comment but subsequently forgot about it :)

> This is actually what we're trying to avoid. If you want to mess with your sensor positions during plotting you should not be using plot_topomap, which (in principle) is meant to make use of a systematic relationship/transformation between actual sensor positions and the subject's head.

Ok, in this sense I agree. I was actually thinking about customizations that make the topomap more accurate (like more realistic head shape and more ellipsoidal shape of channel positions, see for example how EGI channel projections changed between 0.19 and 0.20, previously were ellipsoidal, now are spherical). But that is a distant perspective and the principle of systematic 3d -> 2d projection without user interference is clear. Thanks for clarifying!