mne-tools/mne-python #7107: BUG - mne coreg issues

bloyl, 2019-12-02:
I'm observing odd behavior in `mne coreg` on a clean install on a macbook.

<details>
  <summary>sys.info</summary>

```
(mne-trunk) a483e75df2f6:~ bloyl$  python -c "import mne; mne.sys_info()"
Platform:      Darwin-18.7.0-x86_64-i386-64bit
Python:        3.7.5 (default, Oct 25 2019, 10:52:18)  [Clang 4.0.1 (tags/RELEASE_401/final)]
Executable:    //anaconda3/envs/mne-trunk/bin/python
CPU:           i386: 8 cores
Memory:        16.0 GB

mne:           0.19.2
numpy:         1.17.4 {blas=mkl_rt, lapack=mkl_rt}
scipy:         1.3.1
matplotlib:    3.1.1 {backend=MacOSX}

sklearn:       0.21.3
numba:         0.46.0
nibabel:       2.5.1
cupy:          Not found
pandas:        0.25.3
dipy:          1.0.0
mayavi:        4.7.1 {qt_api=pyqt5, PyQt5=5.13.2}
pyvista:       0.22.4
vtk:           8.1.2
```
</details>

The issue is that head surfaces aren't rendering and points (fids, dig etc) are only sometimes visible depending on how the view is rotated etc.

Interestingly this only happens when called this way. `mne coreg`
If the subjects dir is supplied, `mne coreg -s sample -d ~/mne_data/MNE-sample-data/subjects`,  it seems to work correctly.

Here is a gif showing the issue with `mne coreg`
[image]

and a screenshot of is working with `mne coreg -s sample -d ~/mne_data/MNE-sample-data/subjects`
[image]

hoechenberger, 2019-12-04:
I've witnessed the same behavior on my Mac too using 0.19 and, I believe, `master`.

agramfort, 2019-12-05:
I could not replicate.

can it be a pb with the fsaverage folder you have? in the first case
it's fsaverage and with full command line you use sample

$ python -c "import mne; mne.sys_info()"

Platform:      Darwin-18.7.0-x86_64-i386-64bit

Python:        3.7.5 (default, Oct 25 2019, 10:52:18)  [Clang 4.0.1
(tags/RELEASE_401/final)]

Executable:    /Users/alex/miniconda3/bin/python

CPU:           i386: 4 cores

Memory:        8.0 GB

mne:           0.20.dev0

numpy:         1.17.3 {blas=mkl_rt, lapack=mkl_rt}

scipy:         1.3.1

matplotlib:    3.1.1 {backend=MacOSX}

sklearn:       0.21.3

numba:         0.46.0

nibabel:       2.5.1

cupy:          Not found

pandas:        0.25.3

dipy:          1.0.0

mayavi:        4.7.1 {qt_api=pyqt5, PyQt5=5.13.2}

pyvista:       0.22.4

vtk:           8.1.2

larsoner, 2019-12-05:
I don't get the same problem, but I do get some wacky behaviors in terms of when distances are reloaded and so forth. I can look into this at some point. In the meantime use the workaround of closing and reloading directly with `mne coreg -s whoever -d subjects_dir_wherever`

larsoner, 2020-03-04:
This is a known VTK issue having to do with how they create OpenGL contexts without depth buffers (!) on some systems. Closing as we "just" need to wait for VTK 9 to land.