mne-tools/mne-python #7114: Read montage from a .bvef file

WilliamLabCea, 2019-12-04:
Hi,

I meet some difficulties when I want to load a .bvef (montage files with mne 0.19.2). With previous version of mne (0.18.2) I used 'mne.channels.read_montage' function to load .bvef

With the new function 'mne.channels.read_custom_montage'  mne report an value error:

`Montage_fname = "/MontageAntneuro2.bvef"
...Montage = mne.channels.read_custom_montage(Montage_fname, head_size=None, unit='auto')
raw = mne.io.read_raw_brainvision(vhdr_fname, montage=Montage, misc='auto', scale=1.0, preload=True,
...                                  verbose=None)
Extracting parameters from /media/258884/VERBATIM HD/EEG_Data/EEG_SoundTTC_2sp/Pilote-EEG2_YN_2019-12-04_14-54-58.vhdr...
Setting channel info structure...
Reading 0 ... 481250  =      0.000 ...   939.941 secs...
<input>:2: RuntimeWarning: Limited 1 annotation(s) that were expanding outside the data range.
Traceback (most recent call last):
  File "<input>", line 2, in <module>
  File "/home/258884/anaconda3/envs/NewPycharm/lib/python3.7/site-packages/mne/io/brainvision/brainvision.py", line 825, in read_raw_brainvision
    verbose=verbose)
  File "</home/258884/anaconda3/envs/NewPycharm/lib/python3.7/site-packages/mne/externals/decorator.py:decorator-gen-151>", line 2, in __init__
  File "/home/258884/anaconda3/envs/NewPycharm/lib/python3.7/site-packages/mne/utils/_logging.py", line 90, in wrapper
    return function(*args, **kwargs)
  File "/home/258884/anaconda3/envs/NewPycharm/lib/python3.7/site-packages/mne/io/brainvision/brainvision.py", line 100, in __init__
    _deprecate_montage(self, "read_raw_brainvision", montage)
  File "/home/258884/anaconda3/envs/NewPycharm/lib/python3.7/site-packages/mne/io/utils.py", line 32, in _deprecate_montage
    raw.set_montage(montage, **kwargs)
  File "</home/258884/anaconda3/envs/NewPycharm/lib/python3.7/site-packages/mne/externals/decorator.py:decorator-gen-30>", line 2, in set_montage
  File "/home/258884/anaconda3/envs/NewPycharm/lib/python3.7/site-packages/mne/utils/_logging.py", line 90, in wrapper
    return function(*args, **kwargs)
  File "/home/258884/anaconda3/envs/NewPycharm/lib/python3.7/site-packages/mne/channels/channels.py", line 488, in set_montage
    set_dig=set_dig, raise_if_subset=raise_if_subset)
  File "/home/258884/anaconda3/envs/NewPycharm/lib/python3.7/site-packages/mne/channels/montage.py", line 1545, in _set_montage
    _mnt = _get_montage_in_head(montage)
  File "/home/258884/anaconda3/envs/NewPycharm/lib/python3.7/site-packages/mne/channels/montage.py", line 1356, in _get_montage_in_head
    return transform_to_head(montage.copy())
  File "/home/258884/anaconda3/envs/NewPycharm/lib/python3.7/site-packages/mne/channels/montage.py", line 935, in transform_to_head
    native_head_t = get_ras_to_neuromag_trans(nasion, lpa, rpa)
  File "/home/258884/anaconda3/envs/NewPycharm/lib/python3.7/site-packages/mne/transforms.py", line 650, in get_ras_to_neuromag_trans
    raise ValueError("Points have to be provided as one dimensional "
ValueError: Points have to be provided as one dimensional arrays of length 3.`

Thanks for your help

larsoner, 2019-12-04:
@WilliamLabCea can you upload the `.bvef` file somewhere so we can take a look?

WilliamLabCea, 2019-12-04:
here is the link for the .bvef file

https://drive.google.com/open?id=1aETFbIJKFnlkPJykF8jt_xCubKqFZneg

larsoner, 2019-12-04:
@WilliamLabCea can you see if #7118 or #7119 fixes it for you?

agramfort, 2019-12-04:
@WilliamLabCea <https://github.com/WilliamLabCea> in what coordinate system
are these points? do you use these points as
channel locations for source imaging?

WilliamLabCea, 2019-12-05:
It is a spherical coordinate system for EEG file. I use this file for channel location (no source imaging reconstruction)

agramfort, 2019-12-05:
our other standard montages do not do the job for you?

>

WilliamLabCea, 2019-12-06:
Unfortunately no, the cap is an waveguard64 and doesn't fit with biosemi64 montage

agramfort, 2019-12-08:
@WilliamLabCea this is now fixed in master and in the maint/0.19 branch. You appear in 0.19.3 is a few days

PR #7118 (2019-12-05): MRG, FIX: Allow unknown when setting montage
Closes #7114.

For backward compat, we should allow setting montages with `coord_frame='unknown'` if not all fids are present. Now we do so with a warning.
closed 2019-12-08 by agramfort