mne-tools/mne-python #7115: ENH: Make calculate_chpi_positions public

larsoner, 2019-12-04:
The only reason `mne.chpi._calculate_chpi_positions` is private is that nobody AFAIK has validated that it actually works at least as well as MaxFilter.

- @bloyl have you used `_calculate_chpi_positions` with the Artemis system?
- @AdoNunes you use it in #7096, have you used it much with CTF data?

I could make a push to validate it on some phantom Neuromag data here at UW. Assuming it looks okay we could actually make it public.

bloyl, 2019-12-04:
for artemis I use 
`mne.chi._fit_device_hpi_positions` and `mne.chi._fit_coil_order_dev_head_trans` regularly and am reasonably happy with them. I also don't have another option so when i'm unhappy with the results I generally blame my data quality.

I have tried to use `_calculate_chpi_positions` on artemis data to do motion artifact detection. This has ingeneral not be successful. I think because of gof tolerances, expectations and/or how I populate the `info['hpi_meas']` and other structures from artemis data. It is on my list of things to investigate, but honestly is pretty low/bordering on the not gonna happen section.

In #7096 @AdoNunes  uses `mne.chpi._calculate_head_pos_ctf` which just reads in data from the cHPI channels in the dsfile. In general I'm happy with the results of this.

on my ctf data I can't use `_calculate_chpi_positions` because the HPI frequencies are too high for my sampling. Again if pressed I could see about opening up my sampling to see if I could use `_calculate_chpi_positions` to do the localization. but the CTF reader doesn't read in the needed hpi frequencies so that would need to be fixed as well.

larsoner, 2019-12-04:
> I have tried to use _calculate_chpi_positions on artemis data to do motion artifact detection. This has ingeneral not be successful. I think because of gof tolerances, expectations and/or how I [...]

Do you have data I could test with? I could take a look at fixing these things.

Bonus points if you can do a quick ~10 sec recording of a head (can be foam) moving in some known directions (left-right, forward-back, up-down, rotations?).

> In #7096 @AdoNunes uses mne.chpi._calculate_head_pos_ctf which just reads in data from the cHPI channels in the dsfile. In general I'm happy with the results of this.

Okay we should probably make this public then.

closed 2020-03-04 by larsoner