mne-tools/mne-python #7116: BUG: printing the info on windows date time

dengemann, 2019-12-04:
Yesterday, at the practical MEEG workshop, we made a striking discovery.

On several Windows computers, ```print(raw.info)``` led to an error pointing at a problem with the date fields. We used anonymized data, which may be at the core of the issue as dates and times may have been implausible.

I hope some of the attendees of our workshop can comment on this by providing a traceback of the error.

@agramfort @larsoner

dengemann, 2019-12-04:
Here's a screenshot (thank you Tal!):

[image]

larsoner, 2019-12-04:
What is the `info['meas_date']` here?

agramfort, 2019-12-04:
1941-03-22 11:04:14 GMT

> raw.info['meas_date']
(-908196946, 988669)

larsoner, 2019-12-05:
This was actually fixed already in `master` by #7036

PR #7124 (2019-12-08): MRG, ENH: Use datetime objects for meas_date
~~Closes #7116.~~

~~First step, replicate the failure (hopefully). It passes locally on Linux but hopefully Azure shows the issue.~~

- [x] Use `datetime | None` objects for `info['meas_date']`.
- [x] Use `datetime | None` for `raw.annotations.orig_time`
- [x] Add code for `set_meas_date` that takes care of annotations and such. The anonymization code can use this / be refactored to use it. It only takes `datetime | None` as an argument.
- [x] ~~Extend set_meas_date and/or anonymize functions to kill some of these other dates (such as `meas_id`)~~
 [... cut]
closed 2019-12-05 by larsoner